rjwats/esp8266-react #185: Compilation error with from the filesystem

OuillMac, 2020-07-28:
Hi,

I disabled PROGMEM in the platformio file to test compile by file and I have this error, it is the second parameter of the function serveStatic which poses problem "* fs"

`lib\framework\ESP8266React.cpp: In constructor 'ESP8266React::ESP8266React(AsyncWebServer*)':
lib\framework\ESP8266React.cpp:57:34: error: expected primary-expression before ',' token
   server->serveStatic("/js/", *fs, "/www/js/");
                                  ^
lib\framework\ESP8266React.cpp:58:35: error: expected primary-expression before ',' token
   server->serveStatic("/css/", *fs, "/www/css/");
                                   ^
lib\framework\ESP8266React.cpp:59:37: error: expected primary-expression before ',' token
   server->serveStatic("/fonts/", *fs, "/www/fonts/");
                                     ^
lib\framework\ESP8266React.cpp:60:35: error: expected primary-expression before ',' token
   server->serveStatic("/app/", *fs, "/www/app/");
                                   ^
lib\framework\ESP8266React.cpp:61:42: error: expected primary-expression before ',' token
   server->serveStatic("/favicon.ico", *fs, "/www/favicon.ico");
                                          ^
*** [.pio\build\esp12e\lib9d8\framework\ESP8266React.cpp.o] Error 1`

rjwats, 2020-07-28:
Humm yes, that should be using the ESPFS predefine now.

rjwats, 2020-07-28:
Should be resolved in master now, let me know if you spot any issues

OuillMac, 2020-07-28:
That solved the problem well!

thank you !

closed 2020-07-28 by OuillMac