rjwats/esp8266-react #187: Some problems

OuillMac, 2020-07-29:
I am giving you a little list of problems that I have found, maybe some are not problems but normal behavior, I don't know!

1) When I build my application is that I send in the esp8266 I find the network for the first connection with my smartphone, I am sent to the captive portal but I do not have time to enter my wifi credentials I am automatically redirected to the choice of networks

this is what the terminal returns an interval of 10/15 seconds :

Connecting to WiFi.
Starting software access point
Starting captive portal on 192.168.4.1
WiFi Disconnected. Reason code=15
WiFi connection dropped, stopping NTP.
WiFi Disconnected. Reason code=202
WiFi connection dropped, stopping NTP.

I manage to enter my credentials from my pc because it does not close the internet window for me but I also have the error message in the terminal

2) the first connection should not show all the links of the menu there should only be the wifi page available as long as the wifi connection credentials are not saved
I tried to do it but I cannot access the WiFiStatus data in AppRouting or in MenuAppBar

3) The reset wifi button in the wifi settings does not work, the same for the security settings

here it is for the moment!

I hope this can be fixed, I love this framework that is really great work!

rjwats, 2020-07-30:
Humm...

On your point 1, some devices do disconnect eagerly if they detect they
cannot access the internet - my laptop does this and it's quite annoying. I
have to forget my home wifi network to stop it from disconnecting...
frustrating but I'm not sure of anything the device can do to avoid this,
if you know differently, let me know! Currently the device attempts to
connect as long as it has a non-blank SSID configured, I'm
considering adding a boolean "enabled" flag to the WiFi settings so you can
be explicit about this.

On 2, this is a subjective one. I would argue it's useful to be able to see
most of the settings regardless of the WiFi connection status. You may be
using the device "off grid" where WiFi is not available and using the
Access Point to configure it, this is why the time can be set manually when
NTP or inactive for instance.

Finally, the "Reset" buttons on the forms reset them by reloading the data
from the device. This is not "reset to defaults" but rather "discard my
changes". The "Factory Reset" feature on the system status screen is
designed to restore the default config however, which is different. It's
questionable whether the "Reset" buttons on the forms are useful  - they
all appear to be functioning correctly for me anyway.

proddy, 2020-07-30:
+1 to remove the Reset button