rjwats/esp8266-react #191: Add support to install this framework via Platformio library manager

garung1611, 2020-08-21:
Recently, I am finding a library for Wifi manager and Web OTA. I've tried quite a lot library and finaly I decide to use IotWebConf. But some days ago, I find out this framework. This framework is perfectly match my needs. It's the best framework when compare to other Wifi manager libraries. But because I cannot search this framework in library manager of PlatformIO (I use PlatformIO with VS Code). So, if you can add support to search and install this framework via Platformio library manager, I think many more people will know about it

rjwats, 2020-08-21:
Yes, It could be useful as a standalone WiFi manager for sure, although
it's pretty large compared to other libraries out there.

This had been the plan for a while and there's an issue already open for
using the framework as a library. I just haven't got around to it yet :)

https://github.com/rjwats/esp8266-react/issues/34

rjwats, 2020-08-23:
> Dear rjwats!
> I've a confused thing but I don't know how to PM to ask you. I'm writing
> small code using your framework. From your example, there's only one state
> variable (led_on) for light state. In my code, there are some state
> variables as follow:
> export interface ControllerState {
> isRunning: boolean;
> currentState:string
> targetTemp:number;
> currentTemp:number;
> controlMargin:number;
> runningCycle:number;
> timeFromState:string;
> timeFromCycle:string;
> timeFromProgress:string;
> }
> In above state variables, there's only one variable (isRunning) used to
> regulate the controller. The others are read only state variables. From
> React UI, I can update isRunning to start or stop the controller. And the
> UI will periodic read the other variables to show on the screen. Could you
> show me how i can do it with your framework?
> With the first objective (update isRunning to regulate the controller),
> now, i'm editting LightStateWebSocketController in your example and i got
> error here because I pass only isRunning variable into setData function
> [image: image]
> <https://user-images.githubusercontent.com/69721339/90981985-76d4dd80-e58e-11ea-80f4-b81a1fa7f508.png>
> To by pass this error, i use this way
> [image: image]
> <https://user-images.githubusercontent.com/69721339/90982029-b26fa780-e58e-11ea-874b-ac6a6772d0c0.png>
> Is this way good enough?
> With the second objective (the UI will prediodic read the other variables
> to show on the screen). I think I have two way:
>
>    -
>
>    use javascript setInterval to periodic read state variables (client
>    method)
>    -
>
>    use backend code to periodic send state variables to web socket
>    connected clients (server method)
>    I don't know which one is better and which one is more compatible with
>    your framework

Yes, setData, unlike setState can't take a partial as it replaces the data
held by the RestComponent entirely. Your approach with the spread operator
is fine. I would have ordered it the other way around to avoid possible
errors when modifying this code in future like so:

{...rest, thing: "some value"}.

This way if "rest" contained a property named "thing", it would be
replaced, in this instance with "some value". On a side note, the component
RestComponent is a bit inflexible at times, it was intended to reduce code
duplication in the UI and does so at the expense of flexibility. I may
change this at some point.

There's nothing particularly wrong with your approach with mixing
state/settings there either, it's perfectly valid. Though as a matter of
preference I would have probably had two endpoints to keep it clear what
was state vs settings Only allowing POST on the settings endpoint. Though
with that approach, you have more endpoints of course - so pick your poison
:)

On "polling" vs "websockets", again, both will work. Personally I'd go with
polling unless there was a good reason to have it be event driven (i.e.
updating the client when something changes instead of on an interval
basis). There is a higher memory overhead for WebSockets and the client
code for polling is perfectly easy to implement, like you say, with
setInterval. Just be aware you'll need to hang on to the interval id and
remove it using clearInterval when the component unmounts.

Sounds like a fun project :)

rjwats, 2020-08-23:
Yes, setData, unlike setState can't take a partial as it replaces the data held by the RestComponent entirely. Your approach with the spread operator is fine. I would have ordered it the other way around to avoid possible errors when modifying this code in future like so:

`
{...rest, thing: "some value"}
`

This way if "rest" contained a property named "thing", it would be replaced, in this instance with "some value". On a side note, the component RestComponent is a bit inflexible at times, it was intended to reduce code duplication in the UI and does so at the expense of flexibility. I may change this at some point.

There's nothing particularly wrong with your approach with mixing state/settings there either, it's perfectly valid. Though as a matter of preference I would have probably had two endpoints to keep it clear what was state vs settings Only allowing POST on the settings endpoint. Though with that approach, you have more endpoints of course - so pick your poison :)

On "polling" vs "websockets", again, both will work. Personally I'd go with polling unless there was a good reason to have it be event driven (i.e. updating the client when something changes instead of on an interval basis). There is a higher memory overhead for WebSockets and the client code for polling is perfectly easy to implement, like you say, with setInterval. Just be aware you'll need to hang on to the interval id and remove it using clearInterval when the component unmounts.

Sounds like a fun project :)

garung1611, 2020-08-24:
Thanks rjwats so much. The "rest" approach sounds good. It's absolutely better than current. But  users need to call    **setData{...rest, thing: "some value"}**  or they just call **setData{thing: "some value"}** and the framework will do the rest?
 Your approach "two endpoints" is better than mine. It makes the code more clear and easy to read and maintenance. 
 About "polling" vs "websockets". I prefer the second approach _(use backend code to periodic send state variables to web socket connected clients)_ because follow reason:

- I'm using websocket method for transfering data between client and ESP because using this method, all the client can update when one client changed the state. So even don't need to periodic update the states, I still use this method because above benefit _(all the client can update when one client changed the state)_. And because i've alread used websocket so we can push back updated data to clients easily.

- With "polling" approach, I think it's pretty good with 1 client. But if two or more client connected, the requests to ESP will be double or more. I think, it will create more load for ESP. 

 I upload your example _(original version, not any changed)_ to my ESP8266 . I repeatedly switch on and off the LED and sometimes, ESP was automatically restarted. I don't know why. Is there anyway to see log, callstack or something else to investigate what make ESP restarted?

rjwats, 2020-08-25:
I have seen reset issues when using serial logging during WebSocket
callbacks on the ESP8266. If you open the serial terminal you can get the
trace which you can decode with:

https://github.com/me-no-dev/EspExceptionDecoder

Have you made any changes to the code, or are you seeing this issue when
using the code as is?

garung1611, 2020-08-27:
Yes, I'm using the code as is. But recently, everything seems to be fine. ESP8266 is not restarted anymore so I've not a chance to investigate why it was restarted in the past.
 rjwats, could you do me a favor? I want the guest user have rights to run scan networks and wifi settings in the Wifi Connection. I've digged into your framework, I removed "disabled" code as follow
[image]

With that modification, the tab "scan networks" and "wifi settings" will show when logged in by guest user but when I click to them, the framework redirect me to root page. Instead clicking there tabs, I type the address "http://localhost:3000/wifi/scan", it's still redirect me to root page. How could I grant guest to scan networks and wifi settings?

rjwats, 2020-08-27:
Have you changed the security predicate in the relevant places in the framework code?
The WiFiScanner and WiFiSettingsService use IS_ADMIN (which is the default for any http endpoint unless overridden):

https://github.com/rjwats/esp8266-react/blob/master/lib/framework/WiFiScanner.cpp
https://github.com/rjwats/esp8266-react/blob/master/lib/framework/WiFiSettingsService.cpp 

You can see how the demo project makes the endpoint accessible to all authenticated users by using the IS_AUTHENTICATED instead:

https://github.com/rjwats/esp8266-react/blob/master/src/LightStateService.cpp  

Hope this helps.

garung1611, 2020-08-28:
Thanks rjwats so much. Everything is fine now.

garung1611, 2020-08-30:
Thanks rjwats. With your framework and react, I can quickly create this app like that
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rjwats, 2020-08-30:
Looking good, glad you could make use of it.

garung1611, 2020-09-01:
Dear rjwats!
 I've finished my small project using "websocket" method. Everything is fine as you seen via my screen capture above. Yesterday, I've tried the "polling" method according to your opinion. My code using "polling" method like follow:
[image]

By this way, React UI can frequently update the states from ESP32, but it frequently redraw the component ControllerStateRestControllerForm so it make alitte bit flickering. By "websocket" method, there's no any flickering. "Websocket" method is like updating data using Ajax and "Polling" method is like updating data by refresh the screen. So, rjwats, the flickering issue is because I code wrong way or because the nature of "Polling" way?

> Yes, setData, unlike setState can't take a partial as it replaces the data held by the RestComponent entirely. Your approach with the spread operator is fine. I would have ordered it the other way [...]

rjwats, 2020-09-01:
No, this is because loadData() sets the loading flag. The "flickering" you are seeing is the loading indicator being rendered during the HTTP request. A bit awkward, one of the reasons I'm no longer a fan of the RestController pattern I've implemented there. It's too restrictive.

If you added the following to the loadData function in RestController:

[image]

You could call it with false to supress the loading indicator:

`loadData(false)`

rjwats, 2020-09-01:
I think the RestController pattern I've implemented is a bit restrictive at times. The loading indicator is why you are seeing flickering during the HTTP request. If you add the following to RestController.tsx you can supress the indicator:

[image]

Now you can call loadData(false) to supress the indicator.

garung1611, 2020-09-03:
Dear rjwats, it did not work. after change the loadData function as you said, I've got this error.
[image]
To bypass this error, I've change the code as follow.
[image]
After that, no error occur but the screen is still flickering so I try another way by directly change the loading variable as follow
[image]
With all above modifications, the screen is still flickering :(

garung1611, 2020-09-04:
Already changed as yours, everything works as intended. But in this way, i've to add arrow function ()=> to everywhere call to loadData. Instead this dirty way,I just change loadData in the interface from arrow function into any type, everything still works. Why you dont choose to change loadData to any type. It seems less dirty a little bit
[image]
[image]

rjwats, 2020-09-04:
Using the "any" type is rarely the right thing to do in my experience. I only reserve it for utility functions which can operate defensively on literally "any" value being passed (_.isNull or _.isNumber are good examples) or possibly to provide compatibility with vanilla JavaScript.

In this situation, using "any" would mean you can now pass an object, number or string into the loadData prop. This would result in a runtime error when the callback is attempted, not to mention using "any" in this way defeats the purpose of using TypeScript.

Whilst it requires more code changes, it's definitely better practice to be stricter with the types in general. As loadData should be a function that takes a single, optional boolean argument it should be declared as such.

That being said - do as you wish :)

rjwats, 2020-09-04:
I also noted an error in my example which you may have already noticed, I wasn't calling the loadData function in my callbacks:

[image]

garung1611, 2020-09-05:
> Whilst it requires more code changes, it's definitely better practice to be stricter with the types in general. As loadData should be a function that takes a single, optional boolean argument it should be declared as such.
> 
Yes. I agree with you :)

> I also noted an error in my example which you may have already noticed, I wasn't calling the loadData function in my callbacks:
> 
> [image]
 
On my machine, onClick={()=> this.props.loadData}> works fine. there is no any error.

rjwats, 2020-12-22:
Closing this now, please re-open if you need further help :)

closed 2020-12-22 by rjwats