rjwats/esp8266-react #197: Please test with LITTLEFS for esp32

lorol, 2020-09-08:
Hi @rjwats 

I tested with minor tweak of platformio.ini and ESPFS.h, at least it builds fine with my "hacky" esp32 library here: https://github.com/lorol/LITTLEFS
See at the PIO note at the the bottom of README. 
You can add also to build_flags  -D CONFIG_LITTLEFS_FOR_IDF_3_2 until the core repositories get moved to newer IDF
Appreciate your validation.

Thank you,

raomin, 2021-03-20:
+1
I moved my project to use LITTLEFS with esp8266-react and it's working fine with `CONFIG_LITTLEFS_SPIFFS_COMPAT 1`
LITTLEFS is really more reliable then SPIFFS, worth the effort IMO.

raomin, 2021-03-23:
Ah, one thing I found is that the LittleFS does not unlink (delete) files if they have opened descriptors.
This is an issue with factory reset:
``
Failed to unlink path "/config/apSettings.json". Has open FD.
Failed to unlink path "/config/mqttSettings.json". Has open FD.
Failed to unlink path "/config/ntpSettings.json". Has open FD.
Failed to unlink path "/config/securitySettings.json". Has open FD.
Failed to unlink path "/config/sensorSettings.json". Has open FD.
``

proddy, 2021-03-23:
I discoverd this too, it's by design in LitteFS (see https://github.com/lorol/LITTLEFS/issues/22). I fixed in my fork with code below (sorry should have done a PR).

```C
void FactoryResetService::factoryReset() {
#ifdef ESP32
  /* 
   * Based on LITTLEFS. Modified by proddy
   * Could be replaced with fs.rmdir(FS_CONFIG_DIRECTORY) in IDF 4.2 when platformio catches up
   */
    File root = fs->open(FS_CONFIG_DIRECTORY);
    File file;
    while (file = root.openNextFile()) {
        char * pathStr = strdup(file.name());
        file.close();
        fs->remove(pathStr);
    }
#elif defined(ESP8266)
    Dir configDirectory = fs->openDir(FS_CONFIG_DIRECTORY);
    while (configDirectory.next()) {
        String path = FS_CONFIG_DIRECTORY;
        path.concat("/");
        path.concat(configDirectory.fileName());
        fs->remove(path);
    }
#endif
    RestartService::restartNow();
}
```

rjwats, 2021-03-24:
Thx for the example Proddy, does the strdup need a free in your example (i'm never sure)?

My plan was to wait for the IDF 4.2 release of arduino-esp32 before bothering with adding LITTLEFS support. I wasn't experiencing issues with SPIFFS but I appreciate that others have. Seeing as the 4.2 release may be a way off... might be worth rethinking.

Would probably have to ditch the documentation around manual FS uploads. I'm assuming the PIO tooling doesn't allow for that with a custom FS library - anyone have any experience?

proddy, 2021-03-24:
yeah should be free'd but no point since the ESP is restarted  ;)

FernandoGarcia, 2022-04-03:
Hi!

I'm facing similar issue and I would like to suggest to add some validation here.

The interface is returning a positive message and restarting the controller but the file is not deleted.

[image]

Best regards.