rjwats/esp8266-react #198: Feature Request: support uploading gzip'd firmware binaries

proddy, 2020-09-15:
Espressif's OTA has support for uploading compressed binaries (https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#compression)

If you replace this line in `UploadFirmwareForm.tsx`
```js
<SingleUpload accept="application/octet-stream" onDrop={this.handleDrop} uploading={uploading} progress={progress} onCancel={onCancel} />
```

with

```js
<SingleUpload accept=".bin,.gz" onDrop={this.handleDrop} uploading={uploading} progress={progress} onCancel={onCancel} />
```

you can upload .gz files. There may be some knock-on effects but it seems to work.

rjwats, 2020-09-16:
I see no problem supporting this assuming both platforms support
decompressing the gzipped data. Does it have sensible error behaviour when
you upload random zip files?

proddy, 2020-09-16:
haven't tested. perhaps instead of .gz should be .bin.gz.

proddy, 2020-09-16:
you're right, uploading a bogus .gz bricks the ESP device. I would have expected the espressif's Updater class to validate the firmware.

rjwats, 2020-09-22:
I've tried uploading bad .bin files and it seems to catch them... though i'm not 100% certain i've tried that on both platforms :)

proddy, 2020-09-26:
yes bad .bin files get caught, bad compressed bins don't. I should try this with espota too and see if that is able to detect it.

closed 2021-08-19 by proddy