rjwats/esp8266-react #199: MQTT connection issue only on ESP32

RadViking, 2020-10-13:
I've tried building the default project for esp8266 and esp32, but I'm having issues with the esp32 version. It seems that on the esp32 version, the MQTT doesn't connect to the broker. I get the following status:
[image]

If I'm looking at the broker log in home assistant, I don't see any incoming failed connections either.

MQTT works fine if I compile for esp8266 with the same MQTT settings.

rjwats, 2020-10-13:
I've not seen that issue before, though an update to the SDK or libraries
could have caused something to break of course.

Let me see if I can reproduce the issue.

rjwats, 2020-10-13:
It appears to still be working fine for me on both ESP32 and ESP8266 devices having built a new firmware from the latest SDK. I have tried connecting by both IP address and hostname and both worked for me on both devices. In my case this is a standalone mosquitto instance running on a VM on my network.

Not quite sure what to suggest, I know that the framework does not support TLS or certificate validation so you'll need to be connecting on an insecure port. Though that will be the same for esp8266 of course and you've managed to connect OK on one of those.

Oddly regardless of whether i use invalida credentials or a completly unresolvable domain name the MQTT status service always returns '0' for disconnect_reason instead of one of the many more useful codes it seems to suggest it might return:

[image]

[image]

I'm not sure what to suggest, other than double check everything and maybe try and connect to another MQTT broker as a test.

RadViking, 2020-10-13:
Thank you for the info.

I tried connecting to another broker with the esp32 and that works. Strange that I can't connect to my home assistant broker.

I'm not sure what to try next either.

RadViking, 2020-10-14:
I figured out the problem.

When I enter "homeassistant.home" or the IP address as the MQTT host, it works on the ESP32 and the ESP8266. When I enter "homeassistant.local", it only works on the ESP8266.

It seems like the problem is in resolving the host name.

The fact that the error message is always the same only added to the confusion though.

Thanks again for the help!

tichaonax, 2020-12-06:
yes I had the same issue "hassio.local" does not resolve from the esp32 but does from the esp8266. I even entered a ticket for this a while back. I ended up using the IP address. May be something is different for this class of dns that end with *.local are handled because other "normal" DNSs do work from the esp32

rjwats, 2020-12-07:
Yes, this is multicast DNS - "mDNS".

I think another user had commented that the ".home" extension works on the
ESP32 and ESP8266 where ".local" only works on the ESP8266.

I'm not aware of any API which exposes the configuration of the internal
mDNS resolution. It may be there and I've just not seen it yet.

I know the MDNS singleton may be used for configuring the device's mDNS
responder. The framework could probably deal with this better. At the
moment the OTA service configures the MDNS responder which isn't ideal if
you wanted to configure it or extend it's configuration yourself.

I'll have a dig around the APIs to see if there are any obvious ways of
configuring the mDNS resolver... if you know of any APIs which allow
for this to be configured let me know.