rjwats/esp8266-react #201: Incorrect NTP Time Zone

arthurkomatsu, 2020-11-02:
I am from São Paulo - Brazil. When I set the timezone to America/Sao_Paulo, local time on NTP sets to GMT+3 instead of GMT-3 (e.g. displays 20h00 instead of 14h00 with GMT at 17h00). Looks like some places on America have this behaviour too, such as America/Argentina/Buenos_Aires.

rjwats, 2020-11-02:
Ok, thanks for reporting this - will take a look.

rjwats, 2020-11-02:
This is awkward, and long story short, looks like it's either a problem with the POSIX TZ parser in the underlying firmware or, more likely, a problem with the TZ strings we use.

We take our time zones strings from here:

https://github.com/nayarsystems/posix_tz_db/blob/master/zones.json

In esp8266-react these time zones are stored in the UI code to avoid having to store a large lookup map in the backend:

https://github.com/rjwats/esp8266-react/blob/master/interface/src/ntp/TZ.tsx

Arduino uses the same POSIX time zones as us, over here:

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/TZ.h

Here's an abbrieviated example of how to use it (from the Arduino examples):

```cpp
#ifndef STASSID
#define STASSID "YOUR_SSID"
#define STAPSK  "YOUR_PASSWORD"
#endif

#include <TZ.h>

#define MYTZ TZ_America_Sao_Paulo

#include <ESP8266WiFi.h>
#include <coredecls.h>
#include <PolledTimeout.h>

#include <time.h>
#include <sys/time.h>

extern "C" int clock_gettime(clockid_t unused, struct timespec *tp);

static time_t now;
static esp8266::polledTimeout::periodicMs showTimeNow(5000);

String toISOString(tm* time) {
  char time_string[25];
  strftime(time_string, 25, "%FT%T%z", time);
  return String(time_string);
}

void showTime() {
  now = time(nullptr);
  Serial.print("gmtime:     ");
  Serial.println(toISOString(gmtime(&now)));
  Serial.print("localtime:  ");
  Serial.println(toISOString(localtime(&now)));
}

void time_is_set_scheduled() {
  showTime();
}

void setup() {
  Serial.begin(115200);
  Serial.println("Starting...");

  settimeofday_cb(time_is_set_scheduled);

  // Set the timezone
  configTime(MYTZ, "pool.ntp.org");

  // start network
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(STASSID, STAPSK);
}

void loop() {
  if (showTimeNow) {
    showTime();
  }
}
```

This demonstrates the same behaviour as we are seeing, +3 instead of -3:

[image]

It looks like the correct POSIX string for your location should be: "BRST+3BRDT+2,M10.3.0,M2.3.0". Meaning you have daylight saving and are +3 for some of the year and +2 for the rest. (Is that correct?)

rjwats, 2020-11-02:
Looking at my local linux timezone files, it looks like "America/Sao_Paulo" is supposed to be -3 all year round, and [wikipedia agrees](https://en.wikipedia.org/wiki/List_of_tz_database_time_zones).

So "<-03>3" probably is the correct POSIX string, though for some reason it's being interpreted as +3 when it's being set using configTime.

Not quite sure where to elevate this... I guess I'll raise it over on the [Arduino](https://github.com/esp8266/Arduino) project repo and see if anyone has any suggestions.

arthurkomatsu, 2020-11-02:
Yeah, you are right, running the code from the core yields to the same problem. Strange that no one ever noticed it. Good work raising an issue in the Arduino repo, I will be following.

rjwats, 2020-11-10:
This was fixed in the Arduino codebase by modifing the timezones so they would worth with the underlying library. I have made the same change. Just testing...

closed 2020-11-10 by rjwats