rjwats/esp8266-react #204: Implement JWT expiry - make period configurable

rjwats, 2020-11-14:


rjwats, 2020-11-14:
JWTs issued by the sign in endpoint are currently perpetually valid (though they can be invalidated by changing the JWT secret).

Let's make them have an optional expiry claim and enforce it in the back end. This is a pre-cursor to a "Token Generator" endpoint which an administrative user can use to generate a token which may be used to interface with APIs (IFTTT or another such service)

proddy, 2020-11-14:
looking forward to the Token Generator! In my project I've built a cheap API using parsing the HTTP POST (e.g. ?name1=value1&name2=value2) to fire off events. Which is insecure and nasty.

rjwats, 2020-11-14:
Yes I've wanted this feature for a while... I've been using the guest
account for this purpose but getting a token means hitting the login
endpoint manually with username and password which is not ideal

rjwats, 2020-12-30:
Including and configuring the expiry period (JWT exp claim) configurable will be tricky, though not impossible.

Including and evaluating an expiry claim relies on the RTC being correctly configured (ideally via NTP) so it won't be suitable for everyone's use cases. Worse - if exp claim is blindly set against authentication tokens it will invalidate tokens issued before the device is configured to connect to the interne. The RTC will suddenly jump forward from 01/01/1970 to "now" when NTP synchronises and you'll be kicked out of the UI and forced to re-authenticate.

I think I'm going to split this into 2 stories... a UI/backend feature for generating JWTs for any given user (admin only feature) and then looking at the exp claim and how best to support it.

Perhaps a solution would be to only generate the exp claim if the RTC has been set by NTP. This is something we don't yet track as the NTP sync call-back wasn't previously exposed through the Arduino APIs (it may be now though). This isn't ideal as the device will generate perpetual JWTs until NTP is synchronized but it might be the nicest compromise.

rjwats, 2020-12-30:
Putting this here so I don't forget later. I had it in mind it would be nice to know that NTP is working as expected and that this could help with the JWT behaviour described above. The two Arduino cores are different in what they currently support. ESP8266 exposes a callback so you can observe the sync events:

`cpp
#include <coredecls.h>

// register a callback to observe the NTP sync event
settimeofday_cb(some_callback);
`

There is an equalivant for esp32 (IDK) but it's not exposed through Arduino yet, though this looks like it could be supported with ESP32 Arduino 1.0.5 which should be coming soon:

`cpp
sntp_set_time_sync_notification_cb(some_callback);
`

Ideally I want to know that the time has been set by NTP. Some code I've seen does this by saying: "hey as long the device doesn't think it's the 1970s everything must be fine". However I would like to support showing the last sync time in the UI as well and the callback is really required to support this.

So yes, going to implement this allowing insecure perpetual tokens to be generated to start with (they can always be invalidated en masse by changing the JWT secret if required). Then when 1.5 is out I'll look at improving NTP status reporting and supporting the exp claim for time-limited access tokens.