rjwats/esp8266-react #206: Some question from noob :)

king2, 2020-11-26:
Hi! First of all, sorry for my bad english, I'm not native speaker.

Thank you for excellent lib/framework! It is the best system interface for ESP I've found.  I am new to React, TypeScript and Arduino (my previous projects was on plain ESP SDK), but selected your framework anyway as base for my firmware - just because do not want to make work which was done so well :)

But I have some questions/proposals:

1. I think that names of tabs should be renamed, removing common parts from them - i.e. "NTP STATUS" and "NTP SETTINGS" - to "STATUS" and "SETTINGS" respectively - it have "Network Time" just above these tabs, so it is pretty clear, which status and settings was meant (and same to other system dialogs). This will make tab names shorter, so these names will be shown on one line each (not word-wrapped) - and this will make interface more beautiful :)

2. If I will clone repo to some folder, then will change 'interface/project' and 'src' dirs, I will have changed files (and new files) overriden when I will update framework from repo. Maybe it will be nice to divide framework into two parts - framework itself and example demo project (which will have framework as subproject)? This question connected with next one, in short words "how to develop software based on esp8266-react in a right way without problems with updating?"

3. I want to change something in system dialogs - for example, I want to have OTA, but do not want it to be configurable. I want to hide JWT secret and users page, but still have authentication (without usernames, just password, which can be changed). It is possible to write my own page that will just change passwords, but how to change part of system interface?
By changing system files in 'interface', making diffs and by applying these diffs on each framework update?
Or I have missed something?
I known that some frameworks have possibility to make a folder with custom files, and these files will be taken instead of system files (if such 'override' files exists). Can React do something like this?

4. If you will make i18n support, I can translate interface into russian (it you need this, of course).

rjwats, 2020-11-27:
Thanks for the suggestions and questions, indeed some of this has already been discussed and explored.

> I think that names of tabs should be renamed, removing common parts from them - i.e. "NTP STATUS" and "NTP SETTINGS" - to "STATUS" and "SETTINGS" respectively - it have "Network Time" just above [...]

Yes, possibly, there is some repetition like you say. Quite subjective though - I'm not a designer. I tried this before and the tabs looked a little "anaemic" to my eye - though maybe if they had icons it could work better. Feel free to adapt them how you see fit for your project.

> If I will clone repo to some folder, then will change 'interface/project' and 'src' dirs, I will have changed files (and new files) overriden when I will update framework from repo. Maybe it will be [...]

The addition of the project was to allow users a "safe space" to partition their customizations from the core features: WiFi, NTP, Security etc... 

The example project keeps its C++ code (in /src) and its UI code (in /interface/src/project). These should be concidered as the "extension points" if you want a smooth upgrade when changes are made to the framework.

As this is a "framework" and not a library, you really have to commit to making it your own if you want to make adaptations to its core functionality.  There are no rules, all of the framework code is fair game if you want to modify how it works. You may have to work a bit harder if there are significant updates to the framework should you decide to upgrade in the future.

> I want to change something in system dialogs - for example, I want to have OTA, but do not want it to be configurable. I want to hide JWT secret and users page, but still have authentication [...]
By changing system files in 'interface', making diffs and by applying these diffs on each framework update?
Or I have missed something?

These adaptations would be hard to support through configuration. It sounds like your requirements are slightly different to that which the framework provides out of the box. 

**On a side note**: Note that there is an "admin" role for users which may be able to facilitate some of your requirements. Non-admins can't change settings but can view status and access the "project". If you're requirements don't fit into that paradigm then you'll have to make changes to the framework code (in /lib/framework) as you see fit.

Again, the project section is the "safe space" for making extensions, but make whatever changes you like if you are prepared to deal with merges further down the line :) Most of this is covered in this section of the readme:

https://github.com/rjwats/esp8266-react#developing-with-the-framework

> If you will make i18n support, I can translate interface into russian (it you need this, of course).

Thanks for the offer, I think i18n is worth tacking at some point. I was hoping to find a solution where the core framework's message bundle could be kept separate from the "project" message bundle and haven't found an approach which works particularly smoothly yet. In particular Format.js is pretty huge ~50k IIRC, but it's the best tool to keep the bundles separate.

It's on my TODO list, here are some WIPs using 2 different approaches if you are interested:

https://github.com/rjwats/esp8266-react/tree/i18n-2
https://github.com/rjwats/esp8266-react/tree/i18n

> It will be nice to add some websockets/dynamic behavior to system state page (at least as example for beginners), for example, CPU load and heap fragmentation.

There is a WebSocket example in the project directory which I added for the sake of demonstration.

I was wondering about re-factoring the interface and having global "status info" service to drive information in the App Menu such as icons that show you if the AP/WiFi is active and show you the system time etc.

There's an issue here for that:

https://github.com/rjwats/esp8266-react/issues/104

king2, 2020-11-27:
> I tried this before and the tabs looked a little "anaemic" to my eye - though maybe if they had icons it could work better.
I'm not sure if it possible in React, but I had a project where I made different behavior on PC with wide screen and smartphones (shorter menu texts for displays less than 768p I guess).

> Again, the project section is the "safe space" for making extensions
Is this means that I just should add '/interface/project' folder to .gitingnore?

> In particular Format.js is pretty huge ~50k IIRC, but it's the best tool to keep the bundles separate
What about this approach:
 - all messages in interface modified so it can be easily detected by script (for example, changing "some text" to "~some text~")
 - we have messages files for system texts, and possibility to connect user messages files ('lang' subfolder in interface/project, for example)
 - language can be selected in features
 - preprocessor takes all messages files with selected language, replaces texts in sources before compression (just removing '~' if translate for such text was not found), and we have interface on selected language

Yes, we will have no possibility to change language 'on-fly', but it takes no extra space in firmware (except of length of texts itself), and make possible to change system language by changing just one line (making framework updates simplier).

king2, 2020-11-27:
And another question: build/upload under platformio takes slightly less than 2 minutes on windows 10, i7 16G RAM PC, is this normal?

rjwats, 2020-11-27:
Probably, the npm build can be slow. A full build takes about 30 seconds on
my PC (ryzen 3700x).

If you aren't making changes to tbe interface, disabling the ui build will
speed things up a lot.

Comment out "pre:scripts/build_interface.py" in platformio.ini

king2, 2020-11-28:
What about idea to merge separate source message files into merged/compressed one at preprocessing time with script?
Can this reduce footprint for i18n lib?

rjwats, 2020-11-28:
The idea of selecting which language(s) you want to bundle at build time is a good one. I would argue though that restricting the approach to only support a single language at build time wouldn't work for everyone. I know of some users of the framework who would like to ship a single multi-lingual binary.

There's also the issue of [interpolation](https://www.i18next.com/translation-function/interpolation), [plural forms](https://www.i18next.com/translation-function/plurals) and [formatting](https://www.i18next.com/translation-function/formatting) which the i18n frameworks aim help with which would likely be much more tricky with your suggested appraoch.

To be perfectly honest, I think getting rid of [Moment](https://bundlephobia.com/result?p=moment@2.24.0) would more than make space for a decent i18n library and a few translations on top :)

So this should be top of the list: https://github.com/rjwats/esp8266-react/issues/169 

IMO.

rjwats, 2020-11-28:
Here are examples of other developers' projects which use components/ideas from the framework:

https://github.com/proddy/EMS-ESP
https://github.com/uncodead/BrewUNO

I've tried to keep the framework slim(ish) so it contains features most people will make use of without too much extra stuff weighing it down. In both examples above the developers have followed their desired approaches and added features to the framework to facilitate their project goals.

I18n should be supported by the framework, it's clearly a missing feature. Adding that, along with modernising the UI code to the latest React standards (and reducing the library overhead) will be my next goal.

> I'm not sure if it possible in React, but I had a project where I made different behavior on PC with wide screen and smartphones (shorter menu texts for displays less than 768p I guess).

The Material UI frameworks supports [breakpoints](https://material-ui.com/customization/breakpoints/) which the framework already uses to support smaller screen sizes,

> Is this means that I just should add '/interface/project' folder to .gitingnore?

No need for this. The code in **/src/** and **/interface/src/project** isn't really "core framework" code and should be safe to change with minimal work required to merge later on.

king2, 2020-11-28:
Thank you for answers!

closed 2020-11-28 by king2