rjwats/esp8266-react #210: Missing file? Build problem

tichaonax, 2020-12-11:
<WWWData.h> referenced from /lib/framework/ESP8266React.h
I do not see this file anymore on your lib but. its referenced and it making my build fail. It exists locally as a file on my older repos.

[image]

[image]

rjwats, 2020-12-12:
That file is generated by the interface build process (the prebuild python
step triggers this by calling npm). Its its not checked in.

Double check you haven't disabled this in platformio config by mistake.

rjwats, 2020-12-12:
[image]

It's this step in the platformio build which creates the WWWData.h file - assuming the NPM process runs successfully on your system.

- Verify this line is present in platformio.ini
- Check the build process has no errors when building the interface with npm

rjwats, 2020-12-22:
@tichaonax - I'm assuming you resolved your issues here.. re-open the ticket if you need further help.

closed 2020-12-22 by rjwats

moayyaed, 2022-01-16:
I had exactly this issue during build process, the reason is actually that the npm was not built successfully with last commit ( update CRA and other dependencies #274 ).

# My development environments information is :
OS Name:                   Microsoft Windows 11 Pro
OS Version:                10.0.22000 N/A Build 22000
npm : 6.14.15
node : v14.18.2
tsc : 4.5.4
# The Solutions that worked with me :

# Step 1:
1- run "npm cache clean --force".  
2- delete the "\interface\node_modules" folder .

# Step 2:
 1.  remove "fsevents": "~2.3.2" from package-lock.json in the following lines:

	Line 5094:         "fsevents": "~2.3.2"
	Line 9431:         "fsevents": "^2.3.2"
	Line 13455:         "fsevents": "^2.3.2"
	Line 13817:         "fsevents": "~2.3.2"
	Line 19837:         "fsevents": "~2.3.2",
	Line 22972:         "fsevents": "^2.3.2",
	Line 25720:         "fsevents": "^2.3.2",
	Line 25997:         "fsevents": "~2.3.2"

 2.  build the project again.

# OR
delete package-lock.json file but take care about incompatible dependencies.