rjwats/esp8266-react #211: Relay + button

bajki, 2020-12-22:
Hi, did anyone try to make relay board with push button that activate relay based on selected config and it can update relay state in react?

rjwats, 2020-12-22:
I've not used the this for controlling relays yet, though it sounds fairly trivial based on what you are looking to do. Is it just the single relay?

It sounds like you are looking to have a physical switch control the state in addition to the user interface (do you need MQTT too?) and have them synchronize state. Sounds like a perfect fit for this project. In your state managing object you could have a loop function to service monitor the state of the button and debounce it. Then it can update the state accordingly. Then you have a listener to update the relay, and a WebSocketTxRx instance to broadcast to the UI :)

The demo shares a lot in common with this as it's controlling a boolean state and has an update handler to turn the light on/off when the state changes:

[image]

You can re-use most of this... all you'll need in addition is the ability to control the state manually, there's a section in the docs which talks about this [here.](https://github.com/rjwats/esp8266-react#stateful-service)

If you were to add a loop function to the state service instance, and call it from the main loop of your program. I'm imagining it would look somthing like this:

```cpp
void loop() {
  if ( /* debounce button and detect push event */ ){
    // toggle the relay state    
    update([&](RelayState& state) {
      state.on = !state.on; 
      return StateUpdateResult::CHANGED;
    }, "button");
  }
}
```

You could of course do this from the main loop in main.cpp too as the setting service instances expose the update function publicly:

```cpp

// main.cpp loop function:
void loop() {
  // run the framework's loop function
  esp8266React.loop();

  // service the button
  if ( /* debounce button and detect push event */ ){
    // toggle the relay state    
    relayStateService.update([&](RelayState& state) {
      state.on = !state.on; 
      return StateUpdateResult::CHANGED;
    }, "button");
  }
}
```

rjwats, 2020-12-29:
I hope you managed to get this to work for you, feel free to re-open this ticket if you need further assistance.

closed 2020-12-29 by rjwats