rjwats/esp8266-react #216: How to mix different controllers on one page?

king2, 2021-01-14:
Hello!

I want to make my own page with several blocks on it:
1. variables from ESP, just to show them to user (updateable via websockets)
2. admin password input box and "Save" button (I do not want multiple users, just admin), I think it will be REST
3. firmware versions (static) and "Update firmware" button

Can I mix all this on one page?
If yes, how to do this?
Thank you in advance!

p.s. I wrote comment in #207 (I cannot reopen issue, because it was closed not by me).

rjwats, 2021-01-14:
There's no problem doing this on one page like you suggest.

You can load & save from more than one endpoint as you see fit. I suggest
you look for some React examples on the web to learn more about this.

You don't  have to use the restController component the framework provides
if it doesn't fit your requirements. The only thing you need to remember is
to provide the Authentication token when you make requests to the back end.
This can be done by using the various fetch functions provided in
Authentication.ts.

king2, 2021-01-14:
As component file (as I understand) can export only one default controller, I cannot just mix two or more controllers (each with its own props and states) on one file. I think programmer who familiar with React can do it, but for me this costs learning React from scratch trying to find complicated example in 'React for beginners' pages (actually, I have googled already for similar example, with no luck). :(

Maybe it will be useful to include in example 'project' more example pages, like:
- using two types of controllers on one page (websocket + rest, for example)
- more complicated state, than just one led_on (more variables, with arrays, maybe)?

But I understand that your lib is to be used by React programmers, so I not hoping that you will help me with my task.

Thanks!

rjwats, 2021-01-14:
If the issue you are facing is how to use exports, then this post describes named and default exports pretty nicely:

https://stackoverflow.com/questions/36795819/when-should-i-use-curly-braces-for-es6-import/36796281#36796281

king2, 2021-01-14:
Lucky me!

I found a friend who is (unexpectedly) familiar with React, so at final we got several files each with its own component, then I use such components in one render() function (in another file for page to import all cmoponents into one place) and export this aggregate component to page.

Everything is working now!
Thanks!

rjwats, 2021-01-14:
Sounds like a good solution.

king2, 2021-01-16:
Still have problems mixing controllers.

For example, I have a GSM module connected to ESP and page with:
  - GSM module firmware version (or "not found")
  - RSSI level
  - Provider balance + "Check Balance" button
  - Input with "Balance autocheck period" + "Save" button

So, here I should have websockets for displaying RSSI, and REST for tuning autocheck period.
Firmware version can be got with websockets (just to not make one more endpoint), same with provider balance (maybe this is not a best solution for some reason?)

Here I have 4 child components, but I need to raise state received with websockets to parent, and then give access to this state to other components. **WebSocketController** does **setState()** in **handleMessage**, bit I have not found a handler to update data to parent (without modifying framework core files).

Another approach is to make parent to be **WebSocketController**, so it can propagate its state to all children.
In other words, **WebSocketController** is always a parent which is getting all information needed for its child components, and child components does updates only.

Can you give an advice which approach to choose? Pro/cons/possible problems?

Maybe you can make best example of such 'mixed' using of components and include it to repo?
Something like on one page:
 - [websockets] led switch 
 - [REST] tunable blink count + save button
 - [???] button to start blinking
 - [???] static info like firmware version or something that can be defined at compile time

Pages with such composite content is typical for web-powered devices. I think such example will help many users of your (definitely excellent) framework.

rjwats, 2021-01-16:
At this point you may be better off managing your own state instead of relying on the HOCs ("restController" and "webSocketController") for this. 

The HOCs are provided for convenience where you have a "page per endpoint" pattern. It was aimed at reducing code duplication in the framework. Things are much improved now with React and the interface is due a refresh soon (to make use of state hooks, async/await and other new ECMA/React features), as a result I'm not entirely sure yet whether the HOC pattern for data loading will be used by the framework in a future update (though the components will remain for backwards compatiblity of course).

Using a [context](https://reactjs.org/docs/context.html) for instance would allow you to declare and manage your state in one place, but access and update that state from deeper in the component tree. It sounds like this is what you are after and will avoid the "prop drilling" approach which can get pretty ugly.

So I would suggest:

- Define and manage your state in a context
- Make your own calls to the back-end (the functions in Authentication.ts can help here)
- Expose state and state update functions with the context as required

There are a few examples of using the context in the framework:

The simpler example is the "FeaturesContext" which is [defined here](https://github.com/rjwats/esp8266-react/blob/master/interface/src/features/FeaturesContext.tsx), and [implement here](https://github.com/rjwats/esp8266-react/blob/master/interface/src/features/FeaturesWrapper.tsx):

The UI uses this context to turn features on an off depending on which features are selected at build time. This only talks to a single endpoint, and only reads, but it is an example you have to hand in the framework and it shows how to use context and TypeScript together and may be best to get to grips with before looking at the other example...

The authentication state is managed with a context too, this is more complex as it caters for state which can be updated by the user (logging out for instance):

https://github.com/rjwats/esp8266-react/tree/master/interface/src/authentication

closed 2021-02-03 by king2