rjwats/esp8266-react #223: How to use a Slider component

jaccobezemer, 2021-01-23:
Hello,

I need a Slider to change the speed of a fan and changed the Switch in LightStateWebSocketController.tsx of the demo project into a Slider. Unfortunately I can't get it to work. It seems that the Slider's onChange isn't fired.
This is my code:

  const handleChange = (event: React.ChangeEvent<HTMLInputElement>) => {
    console.log(event.target.value);
    setData({ speed: event.target.valueAsNumber }, saveData);
  }
      <BlockFormControlLabel
        control={
          <Slider
            value={typeof data.speed === 'number' ? data.speed : 0}
            valueLabelDisplay="auto"
            onChange={() => handleChange}
          />
        }
        label="LED State?"
      />

How can I get this to work?

Thanks in advance

rjwats, 2021-01-24:
Sure,

Here's an example of a working slider component:

https://github.com/rjwats/esp8266-react/blob/audio-led2/interface/src/project/LedSettingsForm.tsx

rjwats, 2021-01-24:
In your particular instance, the problem may be that you are using a lambda function as your onChange handler. The lambda will be called with the onChange event and it will return the function you are actually trying to invoke (hence you are observing it not being called)..

The example above may be useful still if this doesn't work, but it's probably as simple as changing waht you are passing into your onChange prop as follows:

```
<Slider
  value={typeof data.speed === 'number' ? data.speed : 0}
  valueLabelDisplay="auto"
  onChange={handleChange}
/>
```

jaccobezemer, 2021-01-24:
Hi rjwats,

Thanks for your reaction.

I'm not a web designer (that's why i'm using your framework;-) so it's not that simple for me:-)
I already tried your suggestion onChange={handleChange} but this results in a compile error: Type '(event: React.ChangeEvent<HTMLInputElement>) => void' is not assignable to type '(event: ChangeEvent<{}>, value: number | number[]) => void'.

And for the working example, I can't factor this in the LightStateWebSocketController.tsx code. I get error: 'this' implicitly has type 'any' because it does not have a type annotation. 

So i'm still kinda lost.

rjwats, 2021-01-24:
Can you send the complete file please?

rjwats, 2021-01-24:
Try the following for your on change function:

```typescript
  const handleChange = (event: React.ChangeEvent<{}>, value: number | number[]) => {
    setData({ speed: value as number }, saveData);
  }
```

Looks like slider can return an array of numbers or a single number, let's tell TypeScript to expect a number here. Here is what my file looks like:

```typescript
import React, { Component } from 'react';
import { ValidatorForm } from 'react-material-ui-form-validator';

import { Slider } from '@material-ui/core';
import { WEB_SOCKET_ROOT } from '../api';
import { WebSocketControllerProps, WebSocketFormLoader, WebSocketFormProps, webSocketController } from '../components';
import { SectionContent, BlockFormControlLabel } from '../components';

import { LightState } from './types';

export const LIGHT_SETTINGS_WEBSOCKET_URL = WEB_SOCKET_ROOT + "lightState";

type LightStateWebSocketControllerProps = WebSocketControllerProps<LightState>;

class LightStateWebSocketController extends Component<LightStateWebSocketControllerProps> {

  render() {
    return (
      <SectionContent title='WebSocket Controller' titleGutter>
        <WebSocketFormLoader
          {...this.props}
          render={props => (
            <LightStateWebSocketControllerForm {...props} />
          )}
        />
      </SectionContent>
    )
  }

}

export default webSocketController(LIGHT_SETTINGS_WEBSOCKET_URL, 100, LightStateWebSocketController);

type LightStateWebSocketControllerFormProps = WebSocketFormProps<LightState>;

function LightStateWebSocketControllerForm(props: LightStateWebSocketControllerFormProps) {
  const { data, saveData, setData } = props;

  const handleChange  = (event: React.ChangeEvent<{}>, value: number | number[]) => {
    setData({ speed: value as number }, saveData);
  }

  return (
    <ValidatorForm onSubmit={saveData}>
      <BlockFormControlLabel
        control={
          <Slider
            value={data.speed }
            valueLabelDisplay="auto"
            onChange={handleChange}
          />
        }
        label="LED State?"
      />
    </ValidatorForm>
  );
}
```

jaccobezemer, 2021-01-24:
Great!
This `const handleChange = (event: React.ChangeEvent<{}>, value: number | number[]) => {
    setData({ speed: value as number }, saveData);
  }`
did the trick.
Many thanks.

rjwats, 2021-01-24:
No worries, happy to help.

closed 2021-01-24 by rjwats