rjwats/esp8266-react #224: ota esp32 does not work

alon24, 2021-01-28:


alon24, 2021-01-28:
I am unable to get OTA working on esp32
I think something is missing in the config, 
Has anyone done that?

rjwats, 2021-01-28:
I have certainly used OTA on both esp32 and esp8266.

Perhaps the platformio configuration settings have changed - will investigate.

rjwats, 2021-01-28:
If you could attach your platformio.ini to this issue that would be great.

rjwats, 2021-01-28:
[image]

I can confirm that it works for me with the following settings (once my device is connected to the network).

Can you confirm which OS you are on, and what errors you are seeing please?

alon24, 2021-01-28:
macos - port is 8266? not 3232?
What about -D NO_GLOBAL_ARDUINOOTA s
Should it be remarked?

```
[platformio]
extra_configs = 
  factory_settings.ini
  features.ini
;default_envs = esp12e
default_envs = node32s

[env]
build_flags=
  ${factory_settings.build_flags}
  ${features.build_flags}
  -D NO_GLOBAL_ARDUINOOTA
  ; Uncomment ENABLE_CORS to enable Cross-Origin Resource Sharing (required for local React development)
  -D ENABLE_CORS
  -D CORS_ORIGIN=\"http://localhost:3007\"
  ; Uncomment PROGMEM_WWW to enable the storage of the WWW data in PROGMEM
  ; -D PROGMEM_WWW

; ensure transitive dependencies are included for correct platforms only
lib_compat_mode = strict

; Uncomment & modify the lines below in order to configure OTA updates
;pload_flags = 
  ;--port=8266 
   ;--auth=esp-react
   
;upload_port = 10.100.102.57  

framework = arduino
monitor_speed = 115200

extra_scripts = 
 pre:scripts/build_interface.py

lib_deps =
  ArduinoJson@>=6.0.0,<7.0.0
  ESP Async WebServer@>=1.2.0,<2.0.0
  AsyncMqttClient@>=0.8.2,<1.0.0
  
[env:esp12e]
platform = espressif8266
board = esp12e
board_build.f_cpu = 160000000L
board_build.filesystem = littlefs

[env:node32s]
; Comment out min_spiffs.csv setting if disabling PROGMEM_WWW with ESP32
;board_build.partitions = min_spiffs.csv
platform = espressif32
board = node32s

```

alon24, 2021-01-28:
Also I cannot figure out how, to upload exactly
2 things - react project and firmware where code is in spiffs

Should it be possible to upload using BOTH ota and usb? (by choosing differnet tasks)?
I do not want to change the platformio.ini file

Also want this options for just the react (in spiffis) and for the rom itslef

alon24, 2021-01-28:
[image]

alon24, 2021-01-28:
so Upload filesystem image OTA fails 

BUT Upload and monitor seems to do OTA

```
Executing task: pio run --target upload --target monitor --environment node32s <

Processing node32s (platform: espressif32; board: node32s; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/node32s.html
PLATFORM: Espressif 32 (2.1.0) > Node32s
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
```

alon24, 2021-01-28:
K, so the OTA issue seems to be a platformio issue
https://github.com/platformio/platform-espressif32/issues/416

alon24, 2021-01-28:
K, closing, found the answers, just did not understnad how to do things
OTA works, the port was the issue.
and platformio will always choose to upload with OTA if it is not remarked out

closed 2021-01-28 by alon24