rjwats/esp8266-react #227: Support to call the configureNTP() on demand

tichaonax, 2021-02-04:
I observed that after running for a number of days the esp8266 time falls behind accurate time. This gets corrected on ip reconfigure or next reboot. Is there a way l can manually call a function to resync NTP service on demand. I have a timer that could do that once every hour or so.

king2, 2021-02-04:
Did you tried calling esp8266React.getNTPSettingsService()->configureNTP()?

@rjwats, BTW, maybe it will be useful also to add field to interface for assigning date and time manually (or button 'Get browser date and time')?

This can be useful for applications where not internet around, but we want date and time to be set.

tichaonax, 2021-02-04:
```
   'class StatefulService<NTPSettings>' has no member named 'configureNTP'
   esp8266React.getNTPSettingsService()->configureNTP();
```

Yes I tried to call the method directly and I get the above error message.

king2, 2021-02-04:
Oh, sorry, indeed configureNTP() is declared as private.

As configureNTP() added as updateHandler for NTPSettingsService, try this as workaround:
`esp8266React.getNTPSettingsService()->update([&](NTPSettings& settings) { return StateUpdateResult::CHANGED; }, "app");`

This should tell StatefulService to propagate updates and call configureNTP() from inside of NTPSettingsService.
Another choice is to call its begin(), but this will lead to re-reading config from FS, what is a little overhead :)

tichaonax, 2021-02-04:
I will that but l did begin() before and its also not accessible same error.

rjwats, 2021-02-05:
Once configured, NTP should be synchronised periodically (hourly by default) on both esp8266 and esp32. You shouldn't have to do anything special to keep it up to date and should only need to reconfigure NTP if you are changing time zone or NTP server.

On esp8266 you can observe the update from sntp by registering a call-back using settimeofday_cb, if you do this you should see updates every hour. ([shown in this example](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino))

I've installed the call-back in mine and set it running over night to see if it's behaving like I'd expect it to. Will report back in the morning. If the NTP isn't being updated automatically then that's an issue - the whole point of the NTP configuration is to keep time synchronised automatically to compensate for the inevitable slipping. 

Here's the first update from my device I just set running:

[image]

If you want to investigate this yourself, it's fairly trivial by including coredecls.h and registering a call-back using settimeofday_cb to observe the synchronization event.

tichaonax, 2021-02-05:
Do you have this functionality in the latest master. I am one version behind probably I did update last month. Let me pull the latest and check. I searched my code I do not see that print statement "Got NTP update..."

rjwats, 2021-02-05:
[image]

Yep, this certainly worked OK for me last night as you can see the NTP automatically refreshed hourly during the night. I can't say why it wouldn't work for you. Other than poor connectivity or some other misconfiguration perhaps?

tichaonax, 2021-02-05:
Thanks for confirming, yes you are right l also verified that it works ok on esp32 but is off by more than 2 minutes on esp8266. Both devices have been on for more than a month no reset.
[image]
[image]

You can see the report from homeassistant both run from 8am to 4pm and scheduled to turn on the pump every 15 minutes. The first one is from esp32 and is out by 1 second confirming NTP sync is working.  The second image from esp8266 is out by more than two minutes. These are test devices in my bedroom and you can hear when the relays turn on and off.

tichaonax, 2021-02-05:
Was your test on esp8266?

rjwats, 2021-02-05:
Yes, I did the test on esp8266 because that's where you reported the issue.
In fact esp32 does not expose a sntp callback yet.

tichaonax, 2021-02-05:
I pulled the latest master and will rebuild that MCU and see. My fall back is to have a controller reboot at midnight of 1st of every month that should at least fix the time. 2 minutes lag a month is not too bad considering l am not docking at the space station where every millisecond matters :-) As usual thanks for your  hard work it really simplified my work especially the integration with home assistant. I have requested a change in homeassistant which they promised to have in their next release. The ability to launch esp8266 UI from homeassistant by clicking on the entity more info. That way l can reconfigure my many devices from homeassistant without having to figure out their ip address.

tichaonax, 2021-02-18:
It works as expected thanks please close as no issue

closed 2021-02-18 by tichaonax