rjwats/esp8266-react #228: FSPersistence - read() to prepare root from return current file content

raomin, 2021-02-13:
Hi @rjwats!
I'm finally moving my old project to a more recent version of esp8266-react (the JS->TS was the hardest part...), I'm glad to see this project is still actively maintained, Congrats and many thanks! 👍 

I'm facing a small issue and I believe this could call for a change: I have a Settings service that allows me to configure a complex object. The _state_ of this object is purely linked to the settings.
So, when `update()` I recreate this object from the new settings and I `_fsPersistence.writeToFS()`.

Now, when `read()` comes I just want to return current file content from `_fsPersistence`. Is there a reason why the framework forces me to serialize in json from my object when the result is already in the file?

I would appreciate that `_fsPersistence` would prepare `JsonObject& root` with the current content of the file and I would still be able to add/modify the json object or not.

What do you think?

rjwats, 2021-02-14:
Hi @raomin,

Glad you could make use of the project :)

The main reason for this is because you may not always want to store the state using file system persistence. For example, if the state is a transient thing, such as a sensor reading or the state of a light. Storing such a value on the file system won't always make sense because it changes frequently and is effectively reset when the device restarts (the example project shows this, with the LightStateService class - no FSPersistence in that class).

You are right that re-serialising the state may seem on the surface to be somewhat wasteful, for example when sending an HTTP response. However the re-serialization will probably be less expensive than fetching from flash because the data must be read from flash storage peripheral chip into RAM anyway before the response can be sent. I don't have any data to back this claim up... but I'd be surprised if this were not the case :)

Are you calling writeToFS() manually? I'm curious because after updating state through a StatefulService instance for which you have created a FSPeristance instance for that should happen automatically because of the update handler which is activated automatically:

[image]

 (I can't see your code and of course there are legit reasons for calling it manually)

I may have misunderstood your question. If so send over a code example so I can get a better grasp of the issue you are raising. I'm certain there are things which can be improved in this area of the framework.

raomin, 2021-02-14:
Thanks @rjwats for this reply, that makes complete sense. 
I definitely understand that some services do not require `FSPersistance` and I completely support it being optional.

Now, in my case, my SettingsServices is essential to manage the complex settings the UI generates (a list of sensors and for each of them, their configuration). I create a complex object from the settings that is used by another class and some of the information in the settings file are actually lost in the final setting object. I simply cannot re-serialize the JSON.
For me, it felt uncoherent that `FSPersistance` would keep the configuration updated every time it changes, loading it at start but would not provide it when you (I) need it. _As for the performance, it is certainly slower to get it from LITTLEFS than ram but I don't think it's a significant difference in the UI._

Now, it is perfectly fine if you don't want to add it in the framework, I just wanted to share my surprise and give a usecase where re-serializing is not possible without persistance. I'll read the settings file manually as `FSPersistance` in not accessible from my settings class (I used same implementation as `LightMqttSettingsService`).

Note _Regarding calling `writeToFS` I didn't see it was an UpdateHandler, so my call is probably superfluous._

Thanks again for your work and support! 👍

raomin, 2021-02-14:
I committed my code, still heavy refactoring in progress, so it might be unclear and possibly misleading.
https://github.com/raomin/esp8266-react/

raomin, 2021-02-15:
Ah I found why I cannot rely on the `writeToFS` handler. 

`read()` from the `HttpPostEndpoint::updateSettings()` is before calling the update handlers (called onDisconnect). 

So, as my read() relies on the settings file it goes like this in my case:
1. `update()` updates the state object
2. `read()` get the JSON from the unmodified FS and update back the state object
3. The unmodified state object is written to the FS by the update() handler
4. I'm upset because I don't get why my new settings does not appear on the UI 😉

I'll keep the manual `writeToFS`

raomin, 2021-02-15:
So, I ended up keep a static jsondoc  with the current state in my config class.
I set the root in `update()` and return it in `read()`.
Works ok, not the smallest footprint though. But the best I could do at this hour.

I believe we can close this one then.

closed 2021-02-15 by raomin