rjwats/esp8266-react #229: PlatformIO : Espressif32 3.1.0 ambiguous overload for "operator=="

justoke, 2021-03-07:
Getting compile errors after updating Platform IO Espressif 32 from 3.0.0 to 3.1.0

[image]

Likely fix would be to use the toString method on IPAddress like so:

[image]

proddy, 2021-03-07:
or just type cast e.g. `... settings.dnsIP1 == (IPAddress)INADDR_NONE ....`

rjwats, 2021-03-08:
Ah yes, I knew this one was coming and probably should have come up with a fix before now. Have pushed a quick fix and dirty fix now...

Perhaps it's best to use a util function rather than the casting or stringifying wherever this particular comparison is done. I'm sure some would suggest inline functions or macros for this sort of thing but I'll raise a PR and we can see what we thing is best.

rjwats, 2021-03-08:
PR attached with a util for isSet and isNotSet for IPAddress, will test and merge tomorrow if I find no issues.

closed 2021-03-09 by rjwats