rjwats/esp8266-react #234: Enable cache on static entries

raomin, 2021-03-20:
Hello @rjwats,

I wanted to suggest to add some cache on all `serveStatic` entries.
`.setCacheControl("max-age=86400")` does get rid of the 7 sec (in my case) for the initial loading of the application.

rjwats, 2021-03-20:
Sounds good to me

rjwats, 2021-03-24:
@raomin I'll get to this this weekend, it's a sensible suggestion. 

Do we think it's worth omitting index.html from the cache though? Otherwise after a deployment you will be forced to do a shift-refresh to pick up the new JavaScript artefacts.

raomin, 2021-04-04:
index.html is very small so yes I would go for omitting it.

Ah because it's next to it, I wanted to suggest something else related to the following issue: When I connect to the captive portal my phone or computer does a lot of requests. Currently it ends up in index.html which, for some services or phone apps, would be parsed and would query the JS files. All this is a massive load for a tiny esp32 on spiffs and often triggers a watchdog timeout.

So I changed a bit the way the index.html is returned to make sure only the browser redirect test get's it:

```c
  server->serveStatic("/favicon.ico", ESPFS, "/www/favicon.ico","max-age=600000");
  server->serveStatic("/index.html", ESPFS, "/www/index.html");
  // Serving all other get requests with "/www/index.htm"
  // OPTIONS get a straight up 200 response
    server->onNotFound([](AsyncWebServerRequest* request) {
    Serial.printf("got 404 request for %s/%s\n", request->host().c_str(), request->url().c_str());
    if (request->method() == HTTP_GET) {
      if (request->host().indexOf("connect") >= 0 || request->host().indexOf("msft") >= 0 ||
          request->host().startsWith("192.168") || request->url().equals("/") ||
          request->url().startsWith("/project") || request->hasHeader("Referer")) {
        request->redirect("/index.html");
      }
      // request->send(ESPFS, "/www/index.html");
    } else if (request->method() == HTTP_OPTIONS) {
      request->send(200);
    } else {
      request->send(404);
    }
  });
```
`192.168` should be replaced by the current AP IP of course...

This limit greatly the amount of requests served.