rjwats/esp8266-react #237: How to listen to other WebSocket Events

BubbleTrouble14, 2021-03-27:
In my application I need to know when the WebSockert Connection closes. 

    case WS_EVT_DISCONNECT:
      Serial.printf("WebSocket client #%u disconnected\n", client->id());
      break;

This is what i need to listen for but sadly my c++ skills arent that good to implement it in the current application. As i would have to edit the WebSocketTxRx class in order to have a sort of callback to my Service Class. I am using your included WebSockets example.

rjwats, 2021-03-27:
Hummm yes - a bit awkward.

I would probably look at introducing a disconnect event callback in
WebSocketConnector as a function pointer (in the framework code like you
said).

Another valid alternative is to write your own websocket behaviour directly
on top of the async server api. I guess it depends on whether you need the
event driven stateful service approach or not :)

If you implement the former approach feel free to raise a PR - I'm not
finding as much time as I'd like to work on this at the moment.

BubbleTrouble14, 2021-03-27:
Okay thx. I'll have a look what i can do and then raise a PR, but I am not that good in c++.

rjwats, 2021-03-27:
No pressure to raise a PR :) I was mainly wanting to give you confidence to
make changes to the framework - it's far from complete feature wise.

I was thinking instead of a disconnect callback, a general purpose event
callback might be better - so any event may be handled. I can't remember
what other events the websocket can raise of the top of my head - just a
thought.

BubbleTrouble14, 2021-03-27:
Yes that would make the most sense. On the React side of things I also had to add some functionally to the Websockets. Cause in my case I just want to send over an object, without looking at its current state. Not sure if its useful for other projects though.

      sendData = (data: any) => {
        const { ws, connected } = this.state;
        if (connected) {
          ws.json(data);
        }
      };

closed 2026-02-09 by rjwats