rjwats/esp8266-react #238: Cannot Build

henzard, 2021-03-28:
`> Executing task: C:\Users\HenzardK\.platformio\penv\Scripts\platformio.exe run --environment node32s <

Processing node32s (platform: espressif32; board: node32s; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Building interface with npm
npm install

up to date, audited 2008 packages in 4s

found 0 vulnerabilities
npm run build

> esp8266-react@0.1.0 build
> react-app-rewired build

Creating an optimized production build...
Compiled successfully.

File sizes after gzip:

The project was built assuming it is hosted at /.
You can control this with the homepage field in your package.json.

The build folder is ready to be deployed.
You may serve it with a static server:

  serve -s build

Find out more about deployment here:

  https://cra.link/deployment

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/node32s.html
PLATFORM: Espressif 32 (3.1.1) > Node32s
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10005.210308 (1.0.5)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.17.3
|-- <ESP Async WebServer> 1.2.3
|   |-- <AsyncTCP> 1.1.1
|   |-- <ArduinoJson> 6.17.3
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.1.1
|-- <framework>
|   |-- <FS> 1.0
|   |-- <ArduinoJson> 6.17.3
|   |-- <ESP Async WebServer> 1.2.3
|   |   |-- <AsyncTCP> 1.1.1
|   |   |-- <ArduinoJson> 6.17.3
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <AsyncTCP> 1.1.1
|   |-- <WiFi> 1.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <AsyncMqttClient> 0.8.2
|   |   |-- <AsyncTCP> 1.1.1
|   |-- <ArduinoOTA> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Update> 1.0
Building in release mode
Linking .pio\build\node32s\firmware.elf
.pio\build\node32s\libFrameworkArduino.a(main.cpp.o): In function `app_main':
C:/Users/HenzardK/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:43: multiple definition of `app_main'
.pio\build\node32s\src\main.c.o:D:\PortableApps\Documents\Projects\ReactNativeEsp32/src/main.c:1: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\node32s\firmware.elf] Error 1
========================================================= [FAILED] Took 51.71 seconds =========================================================

Environment    Status    Duration
-------------  --------  ------------
node32s        FAILED    00:00:51.712
==================================================== 1 failed, 0 succeeded in 00:00:51.712 ==================================================== 
The terminal process "C:\Users\HenzardK\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'node32s'" terminated with exit code: 1.
Terminal will be reused by tasks, press any key to close it.
`

henzard, 2021-03-28:
>C:\Users\HenzardK\.platformio\penv\Scripts\platformio.exe run --environment node32s -v
Processing node32s (board_build.partitions: min_spiffs.csv; platform: espressif32; board: node32s; build_flags: -D FACTORY_WIFI_SSID=\"\", -D FACTORY_WIFI_PASSWORD=\"\", -D FACTORY_WIFI_HOSTNAME=\"#{platform}-#{unique_id}\", -D FACTORY_AP_PROVISION_MODE=AP_MODE_DISCONNECTED, -D FACTORY_AP_SSID=\"WeighsoftEsp-#{unique_id}\", -D FACTORY_AP_PASSWORD=\"password\", -D FACTORY_AP_CHANNEL=1, -D FACTORY_AP_SSID_HIDDEN=false, -D FACTORY_AP_MAX_CLIENTS=4, -D FACTORY_AP_LOCAL_IP=\"192.168.4.1\", -D FACTORY_AP_GATEWAY_IP=\"192.168.4.1\", -D FACTORY_AP_SUBNET_MASK=\"255.255.255.0\", -D FACTORY_ADMIN_USERNAME=\"admin\", -D FACTORY_ADMIN_PASSWORD=\"admin\", -D FACTORY_GUEST_USERNAME=\"guest\", -D FACTORY_GUEST_PASSWORD=\"guest\", -D FACTORY_NTP_ENABLED=true, -D FACTORY_NTP_TIME_ZONE_LABEL=\"Africa/Johannesburg\", -D FACTORY_NTP_TIME_ZONE_FORMAT=\"GMT0BST,M3.5.0/1,M10.5.0\", -D FACTORY_NTP_SERVER=\"time.google.com\", -D FACTORY_OTA_PORT=8266, -D FACTORY_OTA_PASSWORD=\"admin\", -D FACTORY_OTA_ENABLED=true, -D FACTORY_MQTT_ENABLED=false, -D FACTORY_MQTT_HOST=\"test.mosquitto.org\", -D FACTORY_MQTT_PORT=1883, -D FACTORY_MQTT_USERNAME=\"\", -D FACTORY_MQTT_PASSWORD=\"\", -D FACTORY_MQTT_CLIENT_ID=\"#{platform}-#{unique_id}\", -D FACTORY_MQTT_KEEP_ALIVE=60, -D FACTORY_MQTT_CLEAN_SESSION=true, -D FACTORY_MQTT_MAX_TOPIC_LENGTH=128, -D FACTORY_JWT_SECRET=\"#{random}-#{random}\", -D FT_PROJECT=1, -D FT_SECURITY=1, -D FT_MQTT=1, -D FT_NTP=1, -D FT_OTA=1, -D FT_UPLOAD_FIRMWARE=1, -D NO_GLOBAL_ARDUINOOTA, -D CORS_ORIGIN=\"http://localhost:3000\", -D PROGMEM_WWW; lib_compat_mode: strict; framework: arduino; monitor_speed: 115200; extra_scripts: pre:scripts/build_interface.py; lib_deps: ArduinoJson@>=6.0.0,<7.0.0, ESP Async WebServer@>=1.2.0,<2.0.0, AsyncMqttClient@>=0.8.2,<1.0.0)

proddy, 2021-03-28:
Problem is most likely in your `ReactNativeEsp32/src/main.c`. In Arduino you can't have a function called `main()`. Just `setup()` and `loop()`.

henzard, 2021-03-28:
It was that file deleted it and now it works thanks

closed 2021-03-28 by henzard