rjwats/esp8266-react #247: Proposal: debug engine

king2, 2021-06-05:
Sometimes it is needed to use ES8266 serial port for communicating with something external, in such cases debugging can be a problem (there is no debugger with trace possibility, and Serial.print unavailable).

Maybe it will be nice to add something like separate websocket (as feature that can be turned on/off) for debugging purposes:
  - container, or iframe, or separate popup window or whatever on browser side with debug log inside
  - method to add a line from application to be added to log in such debug window

What do you think about this idea?

rjwats, 2021-06-08:
Agree that this would be useful, there's actually an issue for this already and some experimental work has been done in a branch if you are intrested in picking it up :)

https://github.com/rjwats/esp8266-react/issues/123

I'm not really finding much time at the moment to focus on this as work is busy, but if you manage to make decent progress feel free to raise a PR. The difficulty at the time was having a stable solution on the ESP8266 (due to memory constraints/async server), it was working fine on the ESP32 though IIRC.

king2, 2021-06-08:
I'm afraid that this is too complicated task for me (I had experience with programming ESP8266 on plain SDK, so Arduino, C++ and React are new stacks for me, too much new to make this task good). 

Now I'm using just a long variable that can be shown on interface page (placing variable at ESP side and look on it in web-interface). I cannot imagine which memory problem it can have with ~13k of free heap RAM :)

Since I cannot make it by myself, I'll wait until somebody more experienced will complete this and it will be added to main branch. I just dropped an idea (not fresh one as I can see :)

lcsjunior, 2023-02-25:
@king2. You want to generate prints, but instead of Serial it would be in the Browser.
In this case, I've already done something like SSE (Server-Sent Events) and would print in any browser's developer tools.
Do you still need it?

closed 2026-02-09 by rjwats