rjwats/esp8266-react #248: strptime build error within new Arduino Core framework v3.0.0

Lukas-Heiligenbrunner, 2021-06-17:
A build error occurs with the newest Arduino core version v3.0.0 :

```
lib/framework/NTPSettingsService.cpp: In member function 'void NTPSettingsService::configureTime(AsyncWebServerRequest*, ArduinoJson::JsonVariant&)':
lib/framework/NTPSettingsService.cpp:78:15: error: 'strptime' was not declared in this scope; did you mean 'strftime'?
   78 |     char* s = strptime(timeLocal.c_str(), "%Y-%m-%dT%H:%M:%S", &tm);
      |               ^~~~~~~~
      |               strftime

```

This seems to be caused by a disabled prototype. 
To fix this you could define your prototype by your own somewhere:

```
extern "C" char *strptime (const char *__restrict, const char *__restrict, struct tm *__restrict);
```

--> see https://github.com/esp8266/Arduino/issues/8122

Kind regards.

rjwats, 2021-06-19:
> --> see esp8266/Arduino#8122

Ah, thanks for the issue report! Probably worth adding in the prototype for
now like you suggested

Lukas-Heiligenbrunner, 2021-06-21:
This is fixed in https://github.com/esp8266/Arduino/pull/8147 and should be included in the next Arduino Core release.
I doubt if the prototype patch is really necessary.

closed 2021-08-20 by Lukas-Heiligenbrunner