rjwats/esp8266-react #249: MQTT Settings restricts Hostname to 32 characters

justoke, 2021-06-21:
The MQTT settings form validates the Host based on a regex and length check. This has never been an issue until just recently when  I hosted an MQTT broker on an Azure container which resulted in a host name with more than 32 characters.

[image]

[image]

I see no technical reason for the restriction and assume this was probably a reused bit of validation. As this is easily overcome, treat this as an FYI.

rjwats, 2022-03-18:
Yes, I'm not sure why this restriction was put in either.

Hostnames segments (I think they are called 'labels') have a max length of 63 characters, though the full hostname can be up to 253 characters (according to linux manual. I'll update the regex.

closed 2022-03-18 by rjwats

justoke, 2022-03-18:
Thank you for sweeping this one up.