rjwats/esp8266-react #252: Getting static assertion error or 64bit systems

tichaonax, 2021-08-21:
.pio/libdeps/esp12e/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:119:3: error: static assertion failed: To use 64-bit integers with ArduinoJson, you must set ARDUINOJSON_USE_LONG_LONG to 1. See https://arduinojson.org/v6/api/config/use_long_long/
  119 |   ARDUINOJSON_ASSERT_INTEGER_TYPE_IS_SUPPORTED(T);

I see this both on windows 10 and Mac OS. It started recently.

rjwats, 2021-08-22:
I don't get this error myself building the framework as-is. The error seems to sugges toy are using 64 bit longs (long long) and need to add a setting to your build configuration to get the compiler to support it... try adding:

#define ARDUINOJSON_USE_LONG_LONG 1

or to your build flags:

-D ARDUINOJSON_USE_LONG_LONG=1

tichaonax, 2021-08-22:
The bigger problem on a 64bit system is that the framework flash alone is taking 91% and this is just the framework code alone.
[image]
Is there a way not to compile with 64bit.

when I compile with my project it goes to 101% and no linger fits
[image]

tichaonax, 2021-08-22:
Previously on 32bit systems my flash size was 91%

tichaonax, 2021-08-24:
I do not know if anyone is seeing the same but building the project on a 64bit machine results in firmware larger than what can fit on the chip. Is there a way to disable compiling to 64bits. Code previously targeted for esp8266 no longer fits and now can only fit on esp32.