rjwats/esp8266-react #254: esp8266 chip- Framework taking 91% of max firmware size on 64bit systems a jump from about 78% on 32 bit

tichaonax, 2021-08-25:
I am experiencing a new problem possibly due to data types for long and pointers sizes between 32bit and 64bit systems.

This is resulting in a bigger firmware size on 64bit systems leaving no room for the actual project particularly on esp8266 unit

I do not know if there is any optimization that can be done but I am having to stop using the esp8266 and move to esp32 chip which has more memory to circumvent this issue.

Please take a look at https://docs.oracle.com/cd/E19205-01/819-5265/bjami/index.html

to see data size comparison, I think the long and pointer are killing us.

tichaonax, 2021-08-28:
Everything works except the default website is not working. OTA still works as I am able to make updates. I cannot reach http://<IP> it says No webpage was found on that web address.

proddy, 2021-08-28:
Ok so if I get this right, a few weeks ago building on ESP8266 worked fine but now the image size is too large? esp8266-react hasn't been modified so it's most likely the external libraries, including the latest arduino core. If that's the case you can modify the `platformio.ini` to go back and use the previous versions, e.g. `platform = espressif8266@2.6.3` will use the older  arduino core 2.7.4. See if that makes a difference to the image size.

rjwats, 2021-08-28:
> the previous versions, e.g. platform = ***@***.*** will use the

I built master a few days ago and it was up in the 90% range on esp8266....
honestly can't remember if that is normal or not because it's been so long
since I've since i've actively used it.

Poddys suggestion of downgrading the pio libs should help determine I'd
it's a lib change which caused it.

Do you still target 8266 proddy?

proddy, 2021-08-28:
> Do you still target 8266 proddy?

Nope, I'm exclusive ESP32 now. Made the switch a year ago. I was spending too much time tweaking my code to make it fit into the ESP8266's 4MB ram. And an ESP32 is only like 4 euros now. https://github.com/emsesp/EMS-ESP32/tree/dev

tichaonax, 2021-09-11:
It used to be 78% so something did happen.

tichaonax, 2021-09-13:
In June l had a branch that compiled to 78%. Now the same branch compiles to 91% looks there might have been some changes in Arduino libraries or something