rjwats/esp8266-react #260: Question: show connectivity, auto re-connect when offline

proddy, 2021-11-03:
Hi Rick, 

Some of the people using my EMS-ESP project get confused with the System/Restart feature where it says 'Device is restarting', the ESP reboots but the web page is not refreshed. It's a trvial thing, but the same happens with uploading firmware where you need to refresh the page when you think it's done.

So I was thinking what's the best way to check for connectivity to the web server and do retry's when it's offline. I would probably use WebSockets but before I embark on this I'm interested in how you would approach it?

rjwats, 2021-11-03:
Yes,

I suppose I'd poll every few seconds until the device responds, perhaps up
to a timeout of a minute, after which I'd display a message saying the
device is taking longer than expected to restart.

Though it would be nice to go further and have a firmware checksum or
version on that page so you can see what's going on after the flash was
successful.

rjwats, 2021-12-01:
Was this still on your TODO list? Sounds like a good one and I have some ideas about how this might fit in to a UX refresh after the release of the new UI.

proddy, 2021-12-01:
yes, it was. It's kind of an important one for my project and often requested. It'll also fix the annoying issue when after a firmware update the UI is not automatically refreshed (I'm pushing out updates weekly). I was going to base it off this [code](https://dev.to/finallynero/using-websockets-in-react-4fkp). But won't get to it soon as I'm still struggling getting the event source not to skip messages (and reading up on when to use Memo and Callback hooks)

proddy, 2021-12-15:
@rjwats any thoughts on how to trigger a UI refresh after a successful firmware upload? Either via the heartbeat socket check or just hardcoded to trigger after the fileupload. This is an important one for me as I'm pushing out web UX changes every few weeks and my users (which I think could well be a few hundred now) always complain about not immediately seeing the latest UX changes. doh.

rjwats, 2021-12-15:
I would show a spinner while I poll the "features" endpoint probably (because that endpoint is always present regardless of which features are enabled). I'd use a fairly short axios timeout, for a period of up to a couple of mins until the endpoint returns a 200 OK. I think this 

Then I would restart call document.location.reload() or force the document location to '/' on success  to force the reload :)

Happy to knock together a PoC :) - i'm off till 2022!

proddy, 2021-12-15:
Lucky you! If you wouldn’t mind coding a small stub that would be super
helpful. I don’t have any time off this year (new job)!

rjwats, 2021-12-16:
https://youtu.be/bYe4CQ9zeh8

proddy, 2021-12-16:
Awesome! very fancy. So how long did that take you? I like the redirect route at the end. Also while we're (you are) at it is there way to verify the firmware with a CRC check?

rjwats, 2021-12-16:
The verification check is an intresting one isn't it. Possibly generating an sha512 signature for the artefact and verifying that would be the way to go. The signature could be optional, and sent as a header on the file upload, and verified if supplied.

Should be as simple as calling md_update for each segment of the payload, and at then verifying it at the end before calling Update.end(true):

[image]

rjwats, 2021-12-16:
Oh, and it took about an hour, just really needed to write a polling component 👍

proddy, 2021-12-19:
I've been using the firmware upload monitor a lot and can say it works flawlessly, a good 1 hour spent!

rjwats, 2021-12-19:
Nice to hear. It's definitely better to have the feedback in the UI :)