rjwats/esp8266-react #263: Support to configure remote device in the framework "Useful To Have"

tichaonax, 2021-11-30:
In an effort to address the issues on devices having low memory. It would be nice to have configuration of those devices through one that has more memory. For example if I only need one device to turn on and off something on a particular schedule the UI overhead to manage that takes most memory on such devices. Since all communication is via rest APIs it should be possible to do all of that from another device that has the same UI but just tricked to make APIs to the low memory device.

I have been able to do that but I had to add support in the framework which puts me at a disadvantage for framework updates. But as a proof of concept I do not deploy the interface code to the low memory devices. These devices send a Homeassistant config for auto discovery which has URL to the "master" device which runs the UI. The master device can still be used for something else.

#### Homeassistant config example.
```
{"~":"homeassistant/switch/ch1-pin-5/3c71bf320509","name":"North Water Pump : 3c71bf320509","unique_id":"switch-pin-5-3c71bf320509","json_attributes_topic":"~/state","cmd_t":"~/set","stat_t":"~/state","icon":"mdi:water-pump","payload_on":"{\"state\":\"ON\",\"iotAdminUrl\":\"http://192.168.0.172/project/auto/channelOne#?device=192.168.0.153\",\"controlPin\":5,\"channelName\":\"North Water Pump\",\"MAC\":\"3c71bf320509\",\"IP\":\"192.168.0.153\",\"startTime\":\"8:00\",\"endTime\":\"16:00\",\"overrideTime\":\"15m\",\"runEvery\":\"15m\",\"offAfter\":\"3m\"}","payload_off":"{\"state\":\"OFF\",\"iotAdminUrl\":\"http://192.168.0.172/project/auto/channelOne#?device=192.168.0.153\",\"controlPin\":5,\"channelName\":\"North Water Pump\",\"MAC\":\"3c71bf320509\",\"IP\":\"192.168.0.153\",\"startTime\":\"8:00\",\"endTime\":\"16:00\",\"overrideTime\":\"15m\",\"runEvery\":\"15m\",\"offAfter\":\"3m\"}"}
```
This then automatically shows up in HA with a clickable link to the configuration page of the remote device of my scheduler in this case and other attributes of interest.

####  Homeassistant UI
[image]

When you click on the URL it takes you to the master device where you can configure the remote device.

#### Master Device UI
[image]

Changes here are saved to the remote device. This works well but would be even more useful to cover the entire device settings in a way behaving like a you just have the entire remote device UI in-front of you when actually it is not.

The remote defaults are set as part of deployment including the master device IP and Homeassistant IP

rjwats, 2021-12-01:
Serving the UI from another device seems like an intresting premise, at this point it feels almost like a react native app might do a better job, because it would be capable of doing what you are suggesting without a master device to serve the UI.

I'm not sure of the best way this could be made a framework feature. Did you have any thoughts?

proddy, 2021-12-01:
Just upgrade to an ESP32. It's worth the extra 4 USD and all your memory problems will go away. The ESP8266 is almost 7 years old now and just not good for this kind of stuff anymore

tichaonax, 2021-12-01:
> Just upgrade to an ESP32. It's worth the extra 4 USD and all your memory problems will go away. The ESP8266 is almost 7 years old now and just not good for this kind of stuff anymore

I agree in principle but there is a bigger bigger picture here and that his the master/slave features that could end up being made available. Initially of course it was driven by low memory but there could be other uses or UI that you may not want to be on the device itself but can be accessible only through one that has access to the one device that supposedly knows everyone.

Lets look at what could be the benefits of this architecture apart from the one use case that made me think of this.

tichaonax, 2021-12-01:
> Serving the UI from another device seems like an interesting premise, at this point it feels almost like a react native app might do a better job, because it would be capable of doing what you are suggesting without a master device to serve the UI.
> 
> I'm not sure of the best way this could be made a framework feature. Did you have any thoughts?

I do not have better thoughts other than the idea that was inspired by memory limitations. But I do agree with the approach of a react app. One problem I had to hack due to the api calls being made from the UI were the CORS that would go away with your approach.

The integration with homeassistant was also a big win for me from the last time I discussed interfacing HA with you they introduced a feature at my request that includes a clickable link to any URL I pass through the config payload. So now I manage all my devices via HA out of the box without doing any custom changes on it and can turn/on using their UI. Its neat but HA learning curve is too much for the ordinary user, the emphasis seems to be on customization and I do not want to go that route.

The only reason I want it supported on the framework is so that we do not do any custom things to the framework to support missing functionality the UI needs as it then becomes counter to the idea of a framework.

The framework has made it simple for me to get most of the things I needed to do that would be very difficult otherwise. So I want the benefits of new features if they could be added to the framework where possible.

rjwats, 2021-12-02:
I'll be sure to leave CORS as a configurable option in the framework so your federated approach still works.

tichaonax, 2021-12-19:
I was looking at the implementation of local development and realized in essence you are allowing for remote device configuration using the proxy. Is it possible to have the same mechanism in place where instead of running on localhost:3000 with the device proxy to another device. The proxy is currently read from the project config files, is it feasible to start the same mechanism where the host is passed as a parameter the first time you access the URL if the proxy is specified then use that instead of the local device for apis. This will automatically take care of all apis.

e.g
```http://192.168.0.180/<some-path>?proxy=192.168.0.186```
On load of the react app the proxy will be set to the one on the query parameter if present else use the device running the UI as proxy

I looked at the documentation for the http-proxy-middleware and looks like its feasible possible with some changes on the Arduino webserver implementation.

This would solve all my issues I am having to work around and would really be useful.

tichaonax, 2021-12-19:
To get remote configuration to work I only changed one framework file as websocket API implementation is a bit different from the other apis.

This allowed the ws routes to work, on line 59 I check so I can use the correct token for ws 
```interface\src\api\authentication.ts```
[image]

This does not address the framework UI which I am hoping could be addressed by my suggestion above with the http-proxy-middleware. If that works then I can un-do my changes
 
You can take a look at my latest master branch here
https://github.com/tichaonax/esp8266-react-scheduler

tichaonax, 2021-12-22:
update to this, I was able to refactor a few files in the project folder now I am able to have the entire UI including settings etc to update a remote device. Essentially on the first react app load I determine if call is remote configure or not and store relevant information in redux for subsequent use for navigation on that remote device. A login screen is presented as expected on each new device visited so you get a new access token. That is all I really wanted and now all I deploy to the remote device is C++ code only.

A potential approach would be passing the proxy to the axios object but I did not go that route.

closed 2021-12-24 by tichaonax