rjwats/esp8266-react #266: AsyncMQTTClient does not work for SSL enabled brokers

mjangir, 2021-12-03:
The UI doesn't accept port 8883 and 8884 for MQTT as AsyncMQTTClient does not support SSL connections.

https://github.com/marvinroger/async-mqtt-client/issues/265#issuecomment-984371173

rjwats, 2022-03-18:
To my knowledge AsyncTCP on ESP32 does not yet have SSL support:

https://github.com/me-no-dev/AsyncTCP/pull/48

Last time I looked at this it appeared that SSL requires too much RAM be be feasible on the esp8266 with all of the other stuff the framework holds in memory - it needs about 20k of RAM just to to the handshake.

henzard, 2022-03-27:
Why not just use https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/mqtt.html

closed 2026-02-09 by rjwats