rjwats/esp8266-react #268: Allow endpoint to be optionally passed as parameter in useRest.ts

tichaonax, 2021-12-11:
I have five channels that manage different endpoints but do a bunch of common things. To avoid unnecessary code duplication could you please change the interface signature ``` export interface RestRequestOptions<D>```
in file ```\esp8266-react\interface\src\utils\useRest.ts```
for the read method ```read: (endPoint: string) => AxiosPromise<D>;```
to pass an optional endpoint parameter so I can construct the endpoint based on dynamic data.
something similar to: 
```
export interface RestRequestOptions<D> {
  read: (endPoint: string) => AxiosPromise<D>;
  update?: (value: D) => AxiosPromise<D>;
}
```
Then on my api I can do something like this.

```
export function readChannelState(endPoint: string): AxiosPromise<ChannelState> {
  return AXIOS.get(`/${endPoint}`);
}
```
Allowing the same code base to be used from different places. This may require a bit of refactoring on other areas of the framework but I think this will save memory and code duplication. I have about 8 apis for each of my 5 channels. Passing a dynamic parameter would mean a total of 8 apis instead of 40.

By the way the react Hooks you recently added make the codebase much more readable.

rjwats, 2021-12-12:
Intention of the compnent is to serve read/update a single resource at a fixed path. Assuming you mount a component knowing which channel it is going to deal with, it's probably possible to manage what you are looking to do with a function to create a "configured" version of an endpoint function. 

Somthing like this...

On the API side:

```
export const createReadChannelApi = (channelId: string): () => AxiosPromise<ChannelData> => {
  return () => AXIOS.get(`/channel/${channelId}`);
};

export const createUpdateChannelApi = (channelId: string): (channelData: ChannelData) => AxiosPromise<ChannelData> => {
  return (channelData) => AXIOS.post(`/channel/${channelId}`, channelData);
};
```

In the component:

```
...
 const {channelId} = props;
...
 const {
    loadData, saving, data, setData, saveData, errorMessage
 } = useRest<ChannelData>({ read: MyApi.createReadChannelApi(channelId), update: MyApi.createUpdateChannelApi(channelId) });
```

(untested - but you probably get the idea)

rjwats, 2021-12-12:
(you could also happily extend the useRest hook in your project if it suites your needs better :)

tichaonax, 2021-12-12:
> (you could also happily extend the useRest hook in your project if it suites your needs better :)

Thanks. Will try the first approach that is essentially what I want to do but missed something that l saw you did. I think it should work. Will definitely let you know.

tichaonax, 2021-12-12:
> 

I have followed this approach and compiles clean. However the api is repeatedly called and no data comes back as if its not waiting for the async network call. There are no errors in the network tab.

[image]

The form keeps on trying to load the data on line 95. Not sure if there is some race condition, the setData() in ```useRest.ts``` gets the data but it never ends up in the form and so it loops fetching data repeatedly.

[image]

In order to fix it I changed the following dependency in the useRest.ts which was causing it.

[image]

changed to an empty array and now no more loops. Can you please look into it. If it works without that dependency I would rather we put the that fix in place so I do not make changes directly to the framework. Otherwise its now working the way I wanted. Thanks! 
[image]

rjwats, 2021-12-12:
Ah, this is because the load/update  functions are changing on every
render, initialising the two functions with a useCallback will fix it.

tichaonax, 2021-12-12:
> Ah, this is because the load/update functions are changing on every render, initialising the two functions with a useCallback will fix it.
> […](#)

Is this a change I need to make on my side or its changes to the framework code

rjwats, 2021-12-12:
> ends up in the form and so it loops fetching data repeatedly. [image:
> image]

A change you cab make in your component .. will provide an example
shortly... out ATM

rjwats, 2021-12-12:
You could try with your API functions somthing like this:

```javascript
export const createReadChannelApi = (channelId: string): AxiosPromise<ChannelData> => {
  return AXIOS.get(`/channel/${channelId}`);
};

export const createUpdateChannelApi = (channelId: string, channelData: ChannelData): AxiosPromise<ChannelData> => {
  return AXIOS.post(`/channel/${channelId}`, channelData);
};
```

And your component like:

```javascript
  const read = useCallback(() => MyApi.createReadChannelApi(channelId),
    [channelId]
  );
  const update = useCallback(
    (channelData: ChannelData) => MyApi.createUpdateChannelApi(channelId, channelData),
    [channelId]
  );

  const {
    loadData, saving, data, setData, saveData, errorMessage
  } = useRest<ChannelData>({ read, update });
```

rjwats, 2021-12-12:
wrapping the functions in a useCallback will mean they are only re-generated if the dependencies change (channelId in this case)

tichaonax, 2021-12-12:
That worked thanks! I rolled back the change I had made to the framework to unblock me.

closed 2021-12-13 by tichaonax