rjwats/esp8266-react #272: Add support to unsubscribe/unregister a config

tichaonax, 2021-12-17:
I have a unique situation where the config sent to MQTT can be changed by user. This results in a new config and a new device registration in MQTT or Homeassistant. That is exactly what I want to happen and is working as designed. However I want to remove the old config in HA as it does not exist anymore. Homeassistnt will hold on to the old config until you reset HA or HA gives you a warning after a while that it can't reach the device. I want to be able to unsubscribe so that if a new config occurs for the same device it deletes the old one. I looked at the current implementation and it does overlay existing config if they exactly match. Here is my examples when the only change is the control pin on the esp device which results in a new config. What happens I end up seeing two devices in HA but one of them is non-existent anymore. After some time HA will complain that it can't reach the device. I have a lot of use-cases for this particular requirement of reassigning the device control pin.

```
{"~":"homeassistant/switch/ch1-pin-13/ac67b20596ec","name":"Pakaipa : ac67b20596ec","unique_id":"switch-pin-13-ac67b20596ec","json_attributes_topic":"~/state","cmd_t":"~/set","stat_t":"~/state","icon":"mdi:water-pump","payload_on":"{\"state\":\"ON\",\"iotAdminUrl\":\"http://192.168.0.172/project/auto/channelOne#?device=192.168.0.172\",\"controlPin\":13,\"channelName\":\"Pakaipa\",\"MAC\":\"ac67b20596ec\",\"IP\":\"192.168.0.172\",\"startTime\":\"8:00\",\"endTime\":\"18:45\",\"overrideTime\":\"4h\"}","payload_off":"{\"state\":\"OFF\",\"iotAdminUrl\":\"http://192.168.0.172/project/auto/channelOne#?device=192.168.0.172\",\"controlPin\":13,\"channelName\":\"Pakaipa\",\"MAC\":\"ac67b20596ec\",\"IP\":\"192.168.0.172\",\"startTime\":\"8:00\",\"endTime\":\"18:45\",\"overrideTime\":\"4h\"}"}
```

```
{"~":"homeassistant/switch/ch1-pin-14/ac67b20596ec","name":"Pakaipa : ac67b20596ec","unique_id":"switch-pin-14-ac67b20596ec","json_attributes_topic":"~/state","cmd_t":"~/set","stat_t":"~/state","icon":"mdi:water-pump","payload_on":"{\"state\":\"ON\",\"iotAdminUrl\":\"http://192.168.0.172/project/auto/channelOne#?device=192.168.0.172\",\"controlPin\":14,\"channelName\":\"Pakaipa\",\"MAC\":\"ac67b20596ec\",\"IP\":\"192.168.0.172\",\"startTime\":\"8:00\",\"endTime\":\"18:45\",\"overrideTime\":\"4h\"}","payload_off":"{\"state\":\"OFF\",\"iotAdminUrl\":\"http://192.168.0.172/project/auto/channelOne#?device=192.168.0.172\",\"controlPin\":14,\"channelName\":\"Pakaipa\",\"MAC\":\"ac67b20596ec\",\"IP\":\"192.168.0.172\",\"startTime\":\"8:00\",\"endTime\":\"18:45\",\"overrideTime\":\"4h\"}"}
```

What has happened here is that I have the same channel and I just want it to now control a different pin on esp. So I changed from pin 13 to pin 14 but everything is the same. 
I want to be able to "un-config/unsubscribe" ```"~":"homeassistant/switch/ch1-pin-13/ac67b20596ec"```
and replace with ```"~":"homeassistant/switch/ch1-pin-14/ac67b20596ec"```

Here is an example of the duplicates in HA, but the bottom one is the one currently active.
[image]

It would be very useful to expose a method to the framework that one can call to un-configure. If you know a way I can do it without touching the framework please let me know.

rjwats, 2021-12-17:
I feel like I might be missing a little context to help me fully understand what it is you are doing. I'm working on the assumption that you are using HA's auto discovery feature for this and are looking to change the path of the topic you are publishing by publishing a new one and removing the old one?

Well MQTT has no concept of deleting retained topics, however you might find that you can effectively achieve this by posting an empty payload to the topic you are attempting to remove the config HA is using for auto discovery. This post seems to suggest that any way:

https://community.home-assistant.io/t/easiest-way-to-remove-mqtt-device/130466/2

If you stored the previously published topic in a vairable then you could do this when the topic changes (or somthing similar).

Of course that won't help you much if you loose your connection to the broker and then re-configure the device, I would personally try and engineer it so my topic paths change as little as possible to avoid having orphened data in the broker.

tichaonax, 2021-12-17:
> 

Let me look at your suggestion, your understanding of what I intent to do is correct. But take a look at this article is seems to suggested that you can remove it.
https://www.hivemq.com/blog/mqtt-essentials-part-4-mqtt-publish-subscribe-unsubscribe/

Will update you on what I find out.

rjwats, 2021-12-17:
You can "unsubscribe" for sure, pubsub does this:

https://github.com/rjwats/esp8266-react/blob/master/lib/framework/MqttPubSub.h

You can't "unpublish" or delete a topic...

In HA, devices are discovered by its subscription on the "discovery prefix"
path and it looks like the only way to get it to actively forget a device
is to publish an empty payload to the topic :)

Looks easy enough, and I can see how I would make this change to the demo
project.

Let me know how is goes!

tichaonax, 2021-12-17:
You are correct, I cannot unsubscribe. Publishing and empty payload does make it forget. I tested that, please include example in the Demo project someone is bound to come across this. But it works flawlessly. I really appreciate your work. I was able to get the remote device configuration to work for restApi and MQTT without modifying the framework. I made the changes in the user defined AXIOS api that you gave an example before. There is no such interface for web sockets so I had to make a slight change to the authentication file.

When I reach a safe point I will push my changes and see when you can do about the websocket.

closed 2021-12-24 by tichaonax