rjwats/esp8266-react #276: El lector de SVG retorna coordenadas Y en una escala demasiado pequeña

hlavrencic, 2021-12-26:
Ejemplo:  inserto una imagen diagonal y recorre:

NewPos: 2117, 2653
0.00
STANDBY: 2117, 2653
STANDBY: 2117, 2653
NewPos: 2117, 2653
0.00
STANDBY: 2117, 2653
NewPos: 2117, 2653
0.00
STANDBY: 2117, 2653
NewPos: 2117, 2653
0.00
STANDBY: 2117, 2653
NewPos: 294, 583
300.00
STANDBY: 294, 583
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1910,"y":2424}}
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1645,"y":2124}}
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1380,"y":1820}}
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1099,"y":1503}}
{"led_on":false,"brightness":0,"arduinoPosition":{"x":810,"y":1173}}
{"led_on":false,"brightness":0,"arduinoPosition":{"x":533,"y":856}}
{"led_on":false,"brightness":0,"arduinoPosition":{"x":294,"y":583}}
NewPos: 327, 651
291.18
STANDBY: 327, 651
{"led_on":false,"brightness":0,"arduinoPosition":{"x":320,"y":630}}
NewPos: 360, 717
300.00
STANDBY: 360, 717
NewPos: 394, 779
300.00
STANDBY: 394, 779
NewPos: 429, 837
300.00
STANDBY: 429, 837
{"led_on":false,"brightness":0,"arduinoPosition":{"x":429,"y":837}}
NewPos: 464, 891
300.00
STANDBY: 464, 891
NewPos: 500, 937
300.00
STANDBY: 500, 937
NewPos: 536, 978
300.00
STANDBY: 536, 978
NewPos: 573, 1019
300.00
STANDBY: 573, 1019
{"led_on":false,"brightness":0,"arduinoPosition":{"x":556,"y":1000}}
NewPos: 609, 1058
300.00
STANDBY: 609, 1058
{"led_on":false,"brightness":0,"arduinoPosition":{"x":609,"y":1058}}
NewPos: 646, 1096
300.00
STANDBY: 646, 1096
NewPos: 683, 1133
300.00
STANDBY: 683, 1133
NewPos: 719, 1170
300.00
STANDBY: 719, 1170
NewPos: 756, 1207
300.00
STANDBY: 756, 1207
{"led_on":false,"brightness":0,"arduinoPosition":{"x":756,"y":1207}}
NewPos: 793, 1244
300.00
STANDBY: 793, 1244
NewPos: 830, 1280
300.00
STANDBY: 830, 1280
NewPos: 867, 1318
300.00
STANDBY: 867, 1318
{"led_on":false,"brightness":0,"arduinoPosition":{"x":863,"y":1314}}
NewPos: 903, 1355
300.00
STANDBY: 903, 1355
NewPos: 940, 1394
300.00
STANDBY: 940, 1394
NewPos: 976, 1434
300.00
STANDBY: 976, 1434
NewPos: 1013, 1474
300.00
STANDBY: 1013, 1474
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1001,"y":1459}}
NewPos: 1049, 1514
300.00
STANDBY: 1049, 1514
NewPos: 1086, 1553
300.00
STANDBY: 1086, 1553
NewPos: 1123, 1590
300.00
STANDBY: 1123, 1590
NewPos: 1160, 1626
300.00
STANDBY: 1160, 1626
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1123,"y":1590}}
NewPos: 1197, 1660
300.00
STANDBY: 1197, 1660
NewPos: 1234, 1695
300.00
STANDBY: 1234, 1695
NewPos: 1271, 1730
300.00
STANDBY: 1271, 1730
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1259,"y":1720}}
NewPos: 1308, 1765
300.00
STANDBY: 1308, 1765
NewPos: 1344, 1800
300.00
STANDBY: 1344, 1800
NewPos: 1381, 1836
300.00
STANDBY: 1381, 1836
NewPos: 1418, 1871
300.00
STANDBY: 1418, 1871
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1385,"y":1839}}
NewPos: 1455, 1907
300.00
STANDBY: 1455, 1907
NewPos: 1492, 1943
300.00
STANDBY: 1492, 1943
NewPos: 1529, 1978
300.00
STANDBY: 1529, 1978
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1505,"y":1956}}
NewPos: 1566, 2014
300.00
STANDBY: 1566, 2014
NewPos: 1603, 2049
300.00
STANDBY: 1603, 2049
NewPos: 1640, 2084
300.00
STANDBY: 1640, 2084
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1640,"y":2084}}
NewPos: 1677, 2119
300.00
STANDBY: 1677, 2119
NewPos: 1714, 2154
300.00
STANDBY: 1714, 2154
NewPos: 1751, 2187
300.00
STANDBY: 1751, 2187
NewPos: 1788, 2219
300.00
STANDBY: 1788, 2219
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1756,"y":2192}}
NewPos: 1825, 2247
300.00
STANDBY: 1825, 2247
NewPos: 1849, 2224
300.00
STANDBY: 1849, 2224
NewPos: 1886, 2257
300.00
STANDBY: 1886, 2257
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1876,"y":2250}}
NewPos: 1922, 2293
300.00
STANDBY: 1922, 2293
NewPos: 1960, 2316
300.00
STANDBY: 1960, 2316
NewPos: 1997, 2353
300.00
STANDBY: 1997, 2353
{"led_on":false,"brightness":0,"arduinoPosition":{"x":1997,"y":2353}}
NewPos: 2033, 2396
300.00
STANDBY: 2033, 2396
NewPos: 2069, 2443
300.00
STANDBY: 2069, 2443
NewPos: 2105, 2491
300.00
STANDBY: 2105, 2491
NewPos: 2140, 2539
300.00
STANDBY: 2140, 2539
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2109,"y":2495}}
NewPos: 2176, 2588
300.00
STANDBY: 2176, 2588
NewPos: 2212, 2637
300.00
STANDBY: 2212, 2637
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2212,"y":2637}}
NewPos: 2247, 2687
300.00
STANDBY: 2247, 2687
NewPos: 2283, 2737
300.00
STANDBY: 2283, 2737
NewPos: 2318, 2787
300.00
STANDBY: 2318, 2787
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2318,"y":2787}}
NewPos: 2354, 2838
300.00
STANDBY: 2354, 2838
NewPos: 2389, 2893
300.00
STANDBY: 2389, 2893
NewPos: 2424, 2947
300.00
STANDBY: 2424, 2947
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2423,"y":2939}}
NewPos: 2460, 2991
300.00
STANDBY: 2460, 2991
NewPos: 2497, 3027
300.00
STANDBY: 2497, 3027
NewPos: 2533, 3064
300.00
STANDBY: 2533, 3064
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2523,"y":3053}}
NewPos: 2565, 3097
300.00
STANDBY: 2565, 3097
NewPos: 2599, 3152
300.00
STANDBY: 2599, 3152
NewPos: 2634, 3210
300.00
STANDBY: 2634, 3210
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2615,"y":3174}}
NewPos: 2668, 3269
300.00
STANDBY: 2668, 3269
NewPos: 2703, 3325
300.00
STANDBY: 2703, 3325
NewPos: 2738, 3379
300.00
STANDBY: 2738, 3379
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2703,"y":3325}}
NewPos: 2773, 3431
300.00
STANDBY: 2773, 3431
NewPos: 2808, 3484
300.00
STANDBY: 2808, 3484
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2808,"y":3474}}
NewPos: 2843, 3541
300.00
STANDBY: 2843, 3541
NewPos: 2878, 3600
300.00
STANDBY: 2878, 3600
NewPos: 2912, 3659
300.00
STANDBY: 2912, 3659
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2892,"y":3621}}
NewPos: 2947, 3715
300.00
STANDBY: 2947, 3715
NewPos: 2982, 3768
300.00
STANDBY: 2982, 3768
{"led_on":false,"brightness":0,"arduinoPosition":{"x":2982,"y":3768}}
NewPos: 3017, 3819
300.00
STANDBY: 3017, 3819
NewPos: 3052, 3872
300.00
STANDBY: 3052, 3872
NewPos: 3088, 3922
300.00
STANDBY: 3088, 3922
{"led_on":false,"brightness":0,"arduinoPosition":{"x":3084,"y":3915}}
NewPos: 3123, 3971
300.00
STANDBY: 3123, 3971
NewPos: 3159, 4018
300.00
STANDBY: 3159, 4018
NewPos: 3188, 4042
300.00
STANDBY: 3188, 4042
{"led_on":false,"brightness":0,"arduinoPosition":{"x":3180,"y":4037}}
NewPos: 3226, 4030
300.00
STANDBY: 3226, 4030
NewPos: 3264, 4026
300.00
STANDBY: 3264, 4026
NewPos: 3300, 4067
300.00
STANDBY: 3300, 4067
{"led_on":false,"brightness":0,"arduinoPosition":{"x":3285,"y":4050}}
NewPos: 3334, 4127
300.00
STANDBY: 3334, 4127
NewPos: 3353, 4175
237.50
STANDBY: 3353, 4175
NewPos: 3386, 4242
295.52
STANDBY: 3386, 4242

hlavrencic, 2021-12-26:
wrong repo

closed 2021-12-26 by hlavrencic

proddy, 2021-12-26:
Lol!!