rjwats/esp8266-react #282: [Feature request] Add a page to configure a bluetooth server or client.

FernandoGarcia, 2022-03-16:
Hi!

I think could be interesting add a page to configure a bluetooth server or client to use with ESP32.

Just basics information like pin code and device name could be very useful.

An use case could be to make a bridge between bluetooth and MQTT.

Best regards.

FernandoGarcia, 2022-03-16:
Application example: https://www.youtube.com/watch?v=M3qr4PVtuCc

Eisbaeeer, 2022-05-04:
@FernandoGarcia Can you explain me, how to build a simple web-page with react to get e.g. buttons like forward, backward, right, left ? You can ping me dirctly by eisbaeeer@gmail.com . Im a beginner in react and will learn that. Greets

FernandoGarcia, 2022-05-04:
Hi!

Unfortunately I can't help because my react skills is null.

I'm just working in C/C++ code.

For this reason I did this feature request.

Best regards.

Eisbaeeer, 2022-06-28:
Ok, then we are on the same level regarding react. This links seems interesting:
https://www.youtube.com/watch?v=XUXwkT3vVcg
https://www.youtube.com/watch?v=R9n32nxrzug
https://www.youtube.com/watch?v=erWibryA_tE