rjwats/esp8266-react #293: [Question] Add permissions to an user

FernandoGarcia, 2022-04-07:
Hi!

I would like to know if is possible to give some permissions to a guest user without rewrite the interface.

I want to give permission to a guest to connect to a WiFi Network and upload a new firmware and block AP and security tabs.

Thanks in advance.

Best regards

rjwats, 2022-04-09:
Not currently.

The UI is built around the two roles ADMIN and GUEST. The interface has a
built-in understanding of which roles have access to which features.

Granular permissions is a worthwhile feature request. I think there's an
issue about JWT expiry and generating API tokens which seems related.

FernandoGarcia, 2022-04-09:
Ok, thanks for your answer.