rjwats/esp8266-react #300: Error npm start or build  Error: ...API schema.  - options.allowedHosts[0] should be a non-empty string.

benn87, 2022-05-20:
npm start 

the result is.
> esp8266-react@0.1.0 start
> react-app-rewired start

Invalid options object. Dev Server has been initialized using an options object that does not match the API schema.
 - options.allowedHosts[0] should be a non-empty string.

What is it?

rjwats, 2022-05-23:
Not somthing I've seen before... have you done "npm install" or "npm ci" first?

What version of node/npm are you using?

rjwats, 2022-05-23:
ok, thanks

closed 2022-05-23 by rjwats

mareerste, 2024-04-10:
In the root of your ReactApp folder, if you haven't already, create '.env' file and add:
DANGEROUSLY_DISABLE_HOST_CHECK=true

This could be used for development environment but should not be used in production!